pvlib/pvlib-python #1546: Consider creating an FAQ page in the User Guide

kanderso-nrel, 2022-09-06:
**Is your feature request related to a problem? Please describe.**
As with any software package with sufficiently large number of users, some usage questions get asked again and again.  Giving the askers the benefit of the doubt that they did some searching before posting their question, I will vouch that search engines are not always very useful if you don't know the magic terms of art to search for.  

**Describe the solution you'd like**
An FAQ page for things that get asked frequently but might not be found easily anywhere else in our docs.

Off the top of my head, here are some ideas for things it could cover:
- What is pvlib?
- How does pvlib compare to other tools (PVsyst, SAM, et al)
- All I have is GHI, how do I get to POA?
- Can I use PVsyst files (PAN/OND) files with pvlib?
- Why don't my simulation results make sense? (answer: check timezone localization)
- I got a warning like `RuntimeWarning: invalid value encountered in arccos`, what does it mean?
- [your suggestions here]

I'm sure trawling github issues/discussions, stackoverflow, and google group threads would turn up more good things to include.  

It might also be a good idea to explicitly try to speak "plain english" in FAQ entries, at least for the title of each entry.  For example an entry might be better titled "All I have is GHI, how do I get POA?" than something mentioning diffuse components and decomposition models. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

cwhanse, 2022-09-06:
This is a good idea. It would be nice to place the link prominently on the top half of the README page.

AdamRJensen, 2022-09-06:
Great! A question concerning when to use pvlib and when not to use pvlib (e.g., financial calculations) would be a great addition.

PR #1549 (2022-12-02): Create FAQ page in sphinx User Guide
 - [x] Closes #1546
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue  [... cut]
closed 2022-12-02 by kanderso-nrel