pvlib/pvlib-python #1550: Switch from LGTM to GitHub code scanning

AdamRJensen, 2022-09-13:
**Describe the bug**
On pvlib's GitHub landing page (readme file), there's a shield/link to [LGTM](https://lgtm.com/projects/g/pvlib/pvlib-python/alerts?mode=list). While following this link, I noticed the following warning message:

> LGTM.com will be shut down in December 2022 — we recommend that you use GitHub code scanning instead. Read more in [our blog post on the GitHub blog](https://github.blog/2022-08-15-the-next-step-for-lgtm-com-github-code-scanning/).

So sometime before December, we should address this issue by either:
- removing the LGTM shield from the readme file
- switch to GitHub code scanning
- find a third alternative

Upon reading the blog post linked in the message, it seems that the best option might be to just wait:

> **October: help migrate repositories to GitHub code scanning**
We will do our best to help migrate repositories that actively use LGTM.com to flag potential security issues in their pull requests. For those repositories, we will create pull requests that add a [GitHub Actions workflow that runs code scanning](https://docs.github.com/en/code-security/code-scanning/automatically-scanning-your-code-for-vulnerabilities-and-errors/setting-up-code-scanning-for-a-repository#setting-up-code-scanning-manually). Once that configuration file is merged, the repository’s source code (and future pull requests) will be scanned by GitHub code scanning. GitHub code scanning will flag any potential security issues [in pull requests](https://github.blog/changelog/2022-06-02-users-can-view-and-comment-on-code-scanning-alerts-on-the-conversation-tab-in-a-pull-request/) and on the repository’s [security tab](https://docs.github.com/en/code-security/code-scanning/automatically-scanning-your-code-for-vulnerabilities-and-errors/managing-code-scanning-alerts-for-your-repository). Once that’s all working as it should, you can disable the LGTM.com integration.

AdamRJensen, 2023-01-16:
As described above, a PR has been made by GitHub #1600.

kanderso-nrel, 2023-01-17:
I'm not convinced that LGTM/GitHub code scanning is particularly useful for us.  Security vulnerability analysis doesn't seem relevant for pvlib (to me, anyway), and I think its other functionality (linting, static analysis) is redundant with our other CI checks.  I'm also not excited about curating the list of "dismissed alerts".  If it were me I would probably get rid of it for this repository.  But I also haven't looked into it very much and would be happy to learn I am overlooking some nice aspects :)

AdamRJensen, 2023-01-17:
Just as a comment, many of the issues that were flagged by LGTM were recently fixed in https://github.com/pvlib/pvlib-python/pull/1559.

wholmgren, 2023-01-17:
For reference #554 added the LGTM badges. I liked them at the time because they showed that pvlib's code quality was very good and it did find a couple of minor bugs. I've been less than enthusiastic about it since then. I don't have a preference on this, just wanted to point to that context.

AdamRJensen, 2023-02-01:
Closing the PR and issue related to adding GitHub code scanning due to lack of interest from maintainers.

closed 2023-02-01 by AdamRJensen