pvlib/pvlib-python #1564: Add iam function based on Schlick 1994

adriesse, 2022-10-02:
**Is your feature request related to a problem? Please describe.**

Equation 15 in Schlick 1994 gives an efficient approximation equation for surface reflectance as a function of incidence angle and could therefore be used as the basis for an iam function.

**Describe the solution you'd like**
A new function called ```schlick``` in the ```iam``` module.

**Reference**

Schlick, C. An inexpensive BRDF model for physically-based rendering. Computer graphics forum 13 (1994).

adriesse, 2022-10-03:
Extinction in the glass is negligible so we can ignore that part.  The point is that when the original Schlick reflectance function is changed to a normalized iam function, the refractive index (which is only used to determine the normal incidence reflectance) drops out.  Thus to see refractive index reappear in the sky-integrated iam is, well, surprising.

cwhanse, 2022-10-05:
> (Also, github comments [now support mathjax](https://github.blog/2022-05-19-math-support-in-markdown/)!) 

Nice!

adriesse, 2022-10-13:
Please note that I have expanded this feature request to include a function for the diffuse iam factors.  These two functions could in fact be done in a separate PR to form a base layer for FEDIS.

I actually wrote and ran them and compared to schlick integrated with marion and they converge well.

PR #1562 (2022-11-28): Implement iam.schlick, iam.schlick_diffuse
 - [x] Closes #1564
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issu [... cut]
closed 2022-11-28 by kanderso-nrel