pvlib/pvlib-python #1566: Update default solar constant

kdebrab, 2022-10-05:
The default solar constant used in `irradiance.get_extra_radiation` is currently 1366.1 W/m².

Recent work revised the average solar constant to 1361.1 W/m² (see https://www.researchgate.net/publication/324269776_A_reevaluation_of_the_solar_constant_based_on_a_42-year_total_solar_irradiance_time_series_and_a_reconciliation_of_spaceborne_observations).

This new value seems now to be gaining broad acceptance, a.o. in certain standards.

Maybe it's a good time to update also the default value in pvlib to 1361.1 W/m².

kanderso-nrel, 2022-10-05:
> I think this should be an input with a recommended default

I've been frustrated by `irradiance.disc` not exposing the solar constant value as an input, so +1 from me.

Regarding default values: if a model was tuned for a specific value of the solar constant then I'm not sure it makes sense to go against its reference by changing the default.

cwhanse, 2022-10-05:
That function is a convenience wrapper for four extraterrestrial irradiance models, which all differ in how the earth-Sun distance is computed. Two of the models specify a solar constant: 'asce' uses 1367.7 and 'spencer' uses 1366.1. 'pyephem' refers to the ephem python package, which doesn't contain a solar constant that I have found, and 'nrel' refers to SPA and there is no specified solar constant for either. 

So we could make the solar constant an input to this function, but I'd want to specify the values for 'asce' and 'spencer'.

adriesse, 2022-10-13:
@cwhanse The constant and the method to calculate sun-earth distance are independent of each other.

The constant probably only matters for the clear-sky models, where the best value to use may be the one used at the time the model was developed.  Plopping in the new value everywhere would probably create a bit of a bias.

cwhanse, 2022-10-13:
> The constant and the method to calculate sun-earth distance are independent of each other.

I agree, I'm thinking to maintain consistency with the underlying reference for those two extraterrestrial irradiance models which specify a constant.