pvlib/pvlib-python #1578: pvsystem.singlediode passes voltage, not diode voltage, to singlediode.bishop88.

reepoi, 2022-10-31:
**Describe the bug**
When `pvsystem.singlediode` is called with `method = 'brentq'` or `method = 'newton'` and `ivcurve_pnts > 0`, `singlediode.bishop88` is called to calculate the voltage and current of the ivcurve points. `singlediode.bishop88` expects diode voltage, but `pvsystem.singlediode` passes it voltage instead.

**To Reproduce**
Steps to reproduce the behavior:
1. Call with `method = 'brentq'` or `method = 'newton'`,
    ```
    pvlib.pvsystem.singlediode(8, 5e-10, 1, 300, 1.8683643583499283, ivcurve_pnts=100, method='brentq')
    ```
2. The first few voltage values are negative.

**Expected behavior**
The ivcurve points should be on this curve: https://cwhanse.github.io/ivcurves/test_cases.html#case-21

**Versions:**
 - ``pvlib.__version__``: 0.9.3
 - ``pandas.__version__``:  1.4.3
 - python: 3.10

mikofski, 2022-10-31:
The title of this issue is incorrect in my opinion.  I cannot find where in `pvsystem.singlediode` “cell voltages” are passed where diode voltages are expected. Here is the only line where `singlediode.bishope88`
https://github.com/pvlib/pvlib-python/blob/b7768b424d87ff86a863b3565ee708b3470cccbc/pvlib/pvsystem.py#L2874
 and it passes diode voltages as expected and returns both cell voltage & current. 
 
I think you are looking for either `i_from_v` or `v_from_i`

kanderso-nrel, 2022-10-31:
The current behavior as @mikofski describes is consistent with the function's docstring:

https://github.com/pvlib/pvlib-python/blob/b7768b424d87ff86a863b3565ee708b3470cccbc/pvlib/pvsystem.py#L2819-L2824

> The first few voltage values are negative.

Although the most useful part of the IV curve is in the first quadrant, the curve as a whole does extend into quadrant II (and IV), so negative voltages in and of themselves are not physically illegitimate.  Is the objection that the function would be more useful if it only returned points in the first quadrant?

cwhanse, 2022-10-31:
I think in this case I may have given wrong direction to @reepoi. We are attempting to use some precisely computed IV pairs as tests for pvlib single diode equation solutions. We want to supply known voltage and get back the corresponding current. When I first looked, I thought that was the intent of

https://github.com/pvlib/pvlib-python/blob/b7768b424d87ff86a863b3565ee708b3470cccbc/pvlib/pvsystem.py#L2876

On second look, that line is passing voltages that are akin to initial guesses at the cell voltages.

To me it suggests that #418 is still relevant: there are uses when e.g. voltage is known and the corresponding current is desired, and `pvlib.pvsystem.singlediode` doesn't support that. Maybe it's a documentation gap, or as @wholmgren pointed out, an incorrect expectation of `singlediode` raised because of the `ivcurve_pnts` parameter.

mikofski, 2022-10-31:
I think if given specific known voltages and the corresponding currents are desired, then [pvlib.singlediode.bishop88_i_from_v()](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.singlediode.bishop88_i_from_v.html) should be used. IMHO there's not an expectation that [`pvsystem.singlediode()`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.singlediode.html) should return currents given voltages. In fact, AFAICT, voltages isn't even an input to this function:

https://github.com/pvlib/pvlib-python/blob/dd6062afdf86ac113ac53d7a493710f5c79b8935/pvlib/pvsystem.py#L2708-L2710

adriesse, 2022-11-02:
I'm not sure what the best place is for this thought, but perhaps it could be useful if IV curve validation data sets actually included the diode voltage as well.

closed 2022-11-30 by cwhanse