pvlib/pvlib-python #1586: Copyright renewal

cwhanse, 2022-11-07:
The current copyright statement is:

Copyright (c) 2013-2021, Sandia National Laboratories and pvlib python Development Team
All rights reserved.

Two problems: 2021 is past, and "Sandia National Laboratories" isn't a legal entity (neither is the "Development Team.")

Sandia's copyright office is willing to provide the copyright assertion on behalf of pvlib-python. To acknowledge that much of the code does not originate with Sandia, they advise that we change the statement to

Copyright (c) 2013-YYYY, National Technology and Engineering Solutions of Sandia, LLC and Contributors.

@pvlib/pvlib-maintainer would like your opinions here.

wholmgren, 2022-11-08:
Thanks @cwhanse for digging into this. I'm +1 for the proposed change.

cwhanse, 2022-11-08:
That's not the impression I would prefer to give. I don't think Sandia would object to transferring the copyright to NumFocus. 

Interesting, here's numpy's copyright statement:

Copyright (c) 2005-2022, NumPy Developers.

kanderso-nrel, 2022-11-08:
> Copyright (c) 2013-YYYY, National Technology and Engineering Solutions of Sandia, LLC and Contributors.

No objection from me if this is what the lawyers recommended, but for my own understanding, what is the reason to list Sandia separately instead of lumping it in with the rest of the Contributors?  Is it that much of pvlib-python is considered a derivative work of the original MATLAB code (to which Sandia presumably holds copyright)?  If so, we could indicate that history with something like this:

> Copyright (c) 2013-2015, National Technology and Engineering Solutions of Sandia, LLC
> Copyright (c) 2015-YYYY, Contributors

I don't know the relevant years off the top of my head.  The main point is that we might prefer to list copyright years separately for different entities.

kanderso-nrel, 2022-11-08:
> list copyright years separately for different entities

pandas does this, see https://github.com/pandas-dev/pandas/pull/31100#issuecomment-575800533 and https://github.com/pandas-dev/pandas/blob/main/LICENSE#L3-L6

cwhanse, 2022-11-08:
Following pandas' phrasing, I like "pvlib-python open source contributors" more than the other options above.

mdeceglie, 2022-11-08:
Re #1585 - This was my attempt to comply with the original BSD3 license which states "Redistributions of source code must retain the above copyright notice, this list of conditions and the following disclaimer". My thinking was that being incorporated into another BSD3 project (pvlib) meets the "list of conditions" part. Then I added the specific Alliance copyright line to the docstring to comply with the "copyright notice" part. But this is my first time repurposing code from one BSD project to another, so it's very possible this isn't the best/correct approach.

adriesse, 2022-11-08:
I googled a bit and am still unclear about what copyright statements for all or individual parts of pvlib actually achieve.  And in the case of a group copyright, is the idea that the copyright for the whole is shared by all, or that each member of the group has copyright to the part they contributed?  It's probably of no practical relevance, but I'm curious anyway.

As for the Matlab origins of pvlib, I don't think copyright passes on to derivative works.

mikofski, 2022-11-09:
I am not a lawyer, but my understanding of copyright is that it is inherent whenever you publish something publicly. I think it’s actually difficult to give up your copyright. Your work belongs to you, and if your work is plagiarized, then you have the right to pursue damages:
https://en.wikipedia.org/wiki/Copyright

Given those constraints, do we even care? I think anything works work here. I guess I’m not too concerned if someone wants to copy our documentation, but maybe there’s scenarios where we do want to control its use?

If we do want better control over our docs, then I would recommend adopting a Creative Commons license: https://creativecommons.org/

I think the copyright is different than our BSD3 license which mostly protects us the code authors from users of our code trying to receive damages from us. That is my naive understanding. See https://opensource.org/

adriesse, 2022-11-09:
Well, for the most part it doesn't matter since no one is exercising commercial rights, so it's just a question of moral rights, i.e. whether we are individually or collectively recognized as authors.

Technically, listing Sandia *and* contributors seems redundant since Sandia is a contributor.  :)  But I'm ok with any of the suggestions above.

The unanswered question would be whether we want to see a sprinkling of additional alternative copyright declarations in the pvlib code and/or doc strings.

PR #1692 (2023-03-15): update copyright
 - [x] Closes #1586 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Pull request is nearly complete and ready for detailed review.
 - [x] Maintainer: Appropriate GitHub Labels (including `remote-data`) and Milestone are assigned to the Pull Request and linked Issue.
closed 2023-03-15 by kanderso-nrel