pvlib/pvlib-python #1587: [ENH][DOC] add instructions to existing users how to migrate their locar repos to main branch

mikofski, 2022-11-08:
**Is your feature request related to a problem? Please describe.**
see #1584 - we plan to change the name of master branch to main, if we do, links on GH are updated, but users will have to follow directions to update their local repos

**Describe the solution you'd like**
Add a temporary section to the docs? announce on google groups

**Describe alternatives you've considered**
do nothing, just announce on g-group, point users to GH docs

**Additional context**
IMHO, I have dragged my feet too long on this, just need to do it asap. It will be confusing for a few days, then done

mikofski, 2022-11-08:
I'm thinking something on this page: https://pvlib-python.readthedocs.io/en/stable/contributing.html

wholmgren, 2022-11-08:
> do nothing, just announce on g-group, point users to GH docs

This. The only people that will be effected are those that can deal with it with reasonable effort. No need to reinvent GH docs.

mikofski, 2022-11-08:
@wholmgren then I suggest just go ahead and do it now. I'll post something on google groups linking to the discussion or the GH help docs

mikofski, 2022-11-08:
FYI: @pvlib/pvlib-core - I can't make the change, I can't see the settings tab in pvlib

cwhanse, 2022-11-08:
@mikofski simply rename `master` branch to `main`?

mikofski, 2022-11-08:
I believe we can follow the steps here: https://github.com/github/renaming#renaming-existing-branches

[image]

and here: https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-branches-in-your-repository/renaming-a-branch
[image]

[image]

then users do what it says here: https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-branches-in-your-repository/renaming-a-branch#updating-a-local-clone-after-a-branch-name-changes

```bash
$ git branch -m OLD-BRANCH-NAME NEW-BRANCH-NAME
$ git fetch origin
$ git branch -u origin/NEW-BRANCH-NAME NEW-BRANCH-NAME
$ git remote set-head origin -a
$ git remote prune origin
```

mikofski, 2022-11-10:
Here is what GH says when renaming master to main:
[image]
* Will update 36 pull requests targeting this branch across 23 repositories.
* Will update 1 branch protection rule that explicitly targets master.

instructions to update local environments:
```
git branch -m master <BRANCH>
git fetch origin
git branch -u origin/<BRANCH> <BRANCH>
git remote set-head origin -a
```

mikofski, 2022-11-10:
OK, it's done. All contributors please update your local environments following GH instructions.

closed 2022-11-10 by mikofski

mikofski, 2022-11-10:
note replace <BRANCH> with "main" so:
```bash
git branch -m master main
git fetch origin
git branch -u origin/main main
git remote set-head origin -a
git remote prune origin
```

mikofski, 2022-11-10:
@pvlib/pvlib-core  here's a wrinkle, my personal fork is not auto updated, I see this lovely GH message
[image]
so that's another step, to rename master branch in my personal fork as well!

kanderso-nrel, 2022-11-10:
> git fetch origin

Probably most people will have `origin` set to their own fork, right?  I have `kanderso-nrel/pvlib-python` as `origin` and `pvlib/pvlib-python` as `upstream`.  I think people will need to rename the branch in their fork on github before these local commands are applicable.

mikofski, 2022-11-10:
Here's what the renaming process looked like for me, with my mistakes included 😉 😜 
[image]
anyway, not too painful and all good now 👍 😺

mikofski, 2022-11-10:
GH really holding our hands here, saw this on upstream:
[image]

mikofski, 2022-11-10:
@kanderso-nrel  I had one more wrinke, after updating origin and my local environment, I still had this:
[image]

So I reran the GH commands with "upstream" as you suggested 
[image]

now all good
[image]