pvlib/pvlib-python #1592: run_model_from_effective_irradiance() fails with temperature data

pasquierjb, 2022-11-17:
**Describe the bug**
`modelchain.run_model_from_effective_irradiance()` provides inconsistent results if provided with temperature data when `total_irradiance` is not computed beforehand.

This is probably because `_prepare_temperature` that is called by `_run_from_effective_irrad` requires `results.total_irrad` to be computed to calculate `poa`
```python
 poa = _irrad_for_celltemp(self.results.total_irrad,
                                  self.results.effective_irradiance)
``` 
in https://github.com/pvlib/pvlib-python/blob/main/pvlib/modelchain.py#L1713

**To Reproduce**
```python
from pvlib import location
from pvlib import modelchain
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS as PARAMS
from pvlib.iotools import get_pvgis_tmy
from pvlib import pvsystem
from copy import deepcopy
import pandas as pd

lat, lon = 36.084, -79.817
tz = 'UTC'
site_location = location.Location(lat, lon, tz=tz, name='Greensboro, NC')
weather = get_pvgis_tmy(lat, lon, map_variables=True)[0]
    
#tmy_nointerpol = get_tmy(lat, lon)

cec_modules = pvsystem.retrieve_sam('CECMod')
cec_module = cec_modules['Trina_Solar_TSM_300DEG5C_07_II_']
cec_inverters = pvsystem.retrieve_sam('cecinverter')
cec_inverter = cec_inverters['ABB__MICRO_0_25_I_OUTD_US_208__208V_']

temp_model_parameters = PARAMS['sapm']['open_rack_glass_glass']

mount = pvsystem.SingleAxisTrackerMount()

# dc arrays
array = pvsystem.Array(mount=mount,
                       module_parameters=cec_module,
                       temperature_model_parameters=temp_model_parameters)

# create system object
system = pvsystem.PVSystem(arrays=[array],
                           inverter_parameters=cec_inverter)

# ModelChain requires the parameter aoi_loss to have a value. pvfactors
# applies surface reflection models in the calculation of front and back
# irradiance, so assign aoi_model='no_loss' to avoid double counting
# reflections.
mc = modelchain.ModelChain(system, site_location, aoi_model="no_loss", spectral_model='no_loss')
mc2 = deepcopy(mc)

mc.run_model(weather)
data = pd.concat([weather, mc.results.effective_irradiance.to_frame(name='effective_irradiance')])
mc2.run_model_from_effective_irradiance(data)
```

`mc.results.ac.sum()`
562653.4881026025

`mc2.results.ac.sum()`
-1313.9999999999998

**Expected behavior**
`run_model_from_effective_irradiance` is used in Examples such as https://pvlib-python.readthedocs.io/en/stable/gallery/bifacial/plot_bifi_model_mc.html where `total_irradiance` is not computed so I would expect at least a warning to inform that the calculation of cell_temperature will fail and cause the rest of the mc.results to be wrong. 

**Versions:**
 - ``pvlib.__version__``:  0.9.2
 - ``pandas.__version__``: 1.5.0
 - python: 3.9

cwhanse, 2022-11-17:
There may be a bug here, I'll look into the details in a bit.

There's one issue with the example code:

```
data = pd.concat([weather, mc.results.effective_irradiance.to_frame(name='effective_irradiance')])
```

duplicates the index values, because it just pastes the `effective_irradiance` at the bottom of `data`. 

```
data = weather.copy()
data['effective_irradiance'] = mc.results.effective_irradiance
```

results in 

mc2.results.ac.sum() of 562653.4885249813

pasquierjb, 2022-11-18:
@cwhanse You are right, I forgot to specify `axis=1` when concatenating the dataframes:
`data = pd.concat([weather, mc.results.effective_irradiance.to_frame(name='effective_irradiance')], axis=1)`

I was confused because I realised that `mc._assign_total_irrad(data)` assigned an empty array to `mc.results.total_irrad` but I missed out that `mc._irrad_for_celltemp` would assign `poa` to be `mc.results.effective_irradiance` if `poa_global` is missing

Therefore in my exemple as I don't consider any AOI nor spectral losses, the results are the same. 

In the case I consider some losses, I would need to pass the aoi also to get the same results:

```python
from pvlib import location
from pvlib import modelchain
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS as PARAMS
from pvlib.iotools import get_pvgis_tmy
from pvlib import pvsystem
from copy import deepcopy
import pandas as pd

lat, lon = 36.084, -79.817
tz = 'UTC'
site_location = location.Location(lat, lon, tz=tz, name='Greensboro, NC')
weather = get_pvgis_tmy(lat, lon, map_variables=True)[0]

cec_modules = pvsystem.retrieve_sam('CECMod')
cec_module = cec_modules['Trina_Solar_TSM_300DEG5C_07_II_']
cec_inverters = pvsystem.retrieve_sam('cecinverter')
cec_inverter = cec_inverters['ABB__MICRO_0_25_I_OUTD_US_208__208V_']

temp_model_parameters = PARAMS['sapm']['open_rack_glass_glass']

mount = pvsystem.SingleAxisTrackerMount()

# dc arrays
array = pvsystem.Array(mount=mount,
                       module_parameters=cec_module,
                       temperature_model_parameters=temp_model_parameters)

# create system object
system = pvsystem.PVSystem(arrays=[array],
                           inverter_parameters=cec_inverter)

mc = modelchain.ModelChain(system, site_location, aoi_model='physical', spectral_model='no_loss')
mc2 = deepcopy(mc)

mc.run_model(weather)

data = pd.concat(
    [weather, 
     mc.results.effective_irradiance.to_frame(name='effective_irradiance'),
     mc.results.total_irrad[['poa_global']]
    ], axis=1)
mc2.run_model_from_effective_irradiance(data)

print(mc.results.ac.sum(), mc2.results.ac.sum())
```
559666.5146782668 559666.5146782668

closed 2022-11-18 by pasquierjb