pvlib/pvlib-python #1593: Is azimuth to be specified from true or magnetic north?

ertong, 2022-11-17:
I'm playing with FixedMount array model and trying to optimize tilt and azimuth. It should be a standard task  and the results should be predictable.

I expect to have the optimal azimuth to be 180, but receive 172.4 ... exactly the difference between true and magnetic pole in my location.

So it looks like, the library is designed for magnetic north ... but it looks like it is better to have true one.

Anyway, I could not find anywhere in the docs which north is used. 

In any case, I think, it should be explicitly stated and consistent throughout the library.

ertong, 2022-11-17:
Ok, thanks for clarification.

> that could be clarified in docstrings.
So, let's leave this issue about docs.

kanderso-nrel, 2022-11-17:
7.6 degrees of azimuth is suspiciously close to 360 * (30 / 1440) = 7.5, a rough estimate of the degrees of azimuth the sun traverses in half an hour.  Perhaps you are using hourly data and calculating timestamps at the edge of each hourly interval rather than the center?

mikofski, 2022-11-17:
@ertong if you are happy would you consider closing this issue? thanks :) Also consider posting questions either in [GH discussions](https://github.com/pvlib/pvlib-python/discussions) or the [google group](https://groups.google.com/g/pvlib-python). Thanks for your feedback :)

cwhanse, 2022-11-17:
I think this issue asks to clarify that north means true north, in docstrings.

ertong, 2022-11-17:
The code snippt below. (Probably, should have posted it in the first place)
(location is changed, so the results are a bit different)

```python
import numpy as np
import pvlib

location = pvlib.location.Location(49.38,  26.84, 'Etc/GMT+2', 304)

weather = pvlib.iotools.get_pvgis_tmy(location.latitude, location.longitude, map_variables=True)[0]

sapm_inverters = pvlib.pvsystem.retrieve_sam('cecinverter')
module = pvlib.pvsystem.retrieve_sam('CECMod')['LONGi_Green_Energy_Technology_Co___Ltd__LR6_72HBD_385M']
inverter = sapm_inverters['ABB__MICRO_0_25_I_OUTD_US_208__208V_']
temperature_model_parameters = pvlib.temperature.TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']

def fixed(tilt, azimuth):
    mount = pvlib.pvsystem.FixedMount(tilt, azimuth)
    array = pvlib.pvsystem.Array(
            mount=mount,
            module_parameters=module,
            temperature_model_parameters=temperature_model_parameters,
            modules_per_string = 2,
            strings = 2,
        )
    system = pvlib.pvsystem.PVSystem(arrays=[array], inverter_parameters=inverter)
    mc = pvlib.modelchain.ModelChain(system, location, aoi_model="no_loss", spectral_model="no_loss")
    mc.run_model(weather)
    S = mc.results.dc.p_mp.sum()
    print(tilt, azimuth, S)
    return S

from scipy.optimize import minimize

x0 = np.array([65, 180])
res = minimize(lambda x: -fixed(x[0],x[1]), x0, method='nelder-mead', bounds=[(0,90),(90, 270)],
               options={'xatol': 1e-1, 'disp': True})
print(res)
```

The result is 32.96522894 tilt and 171.6903322 azimuth.

On the same time, different online tools give different optimal orientation. For example, https://re.jrc.ec.europa.eu/pvg_tools/en/tools.html for the same point recommends 35 tilt and 179 (-1 out of south).

I understand, that it is incorrect to compare different tools in a such manner due to a lot different models used. But I expected the difference to be one order of magnitude lower.

ertong, 2022-11-17:
>I think this issue asks to clarify that north means true north, in docstrings.
Yes, the other things, are out of scope here.

cwhanse, 2022-11-17:
> But I expected the difference to be one order of magnitude lower.

I would guess that the change in annual energy is rather small in the neighborhood of the returned optimum tilt and azimuth, so perhaps different transposition models etc. and TMY data could cause that variation in optimal orientation.

adriesse, 2022-11-28:
Since this is not a common source of confusion, I don't really think the docs *need* to be changed, but I would not oppose anyone doing so.

steve-ransome, 2022-12-04:
I have no comments on this but saw this "interesting" article on magnetic, true and grid north (which may vary from country to country). https://www.ordnancesurvey.co.uk/newsroom/blog/magnetic-true-grid-north-align-over-great-britain

matsuobasho, 2023-11-10:
Happy to tackle this issue to close it out, since labeled as 'good first issue'.  Is the recommended course of action here to find where north is described in docstrings and to add a clarification that it refers to true north?

ertong, 2023-11-11:
From my point of view, clarifying docstrings - is ok.
Anyway, this issue by itself remains searchable in case of confusion.

closed 2024-05-14 by AdamRJensen