pvlib/pvlib-python #1594: Create new temperature model(s) incorporating a simple radiative loss term.

adriesse, 2022-11-17:
**Is your feature request related to a problem? Please describe.**
The most frequently used operating temperature models do not account properly for radiative losses to the sky.

**Describe the solution you'd like**
Create enhanced versions of the existing models (at least one) incorporating a simple but effective radiative loss term as described in: Driesse, A. et al (2022) "Improving Common PV Module Temperature Models by Incorporating Radiative Losses to the Sky". SAND2022-11604. 

**Describe alternatives you've considered**
Watching TV.

cwhanse, 2022-11-17:
The point in the technical report is that temperature models fit measurements better when a radiative loss term is included. Is that correct?

The generic coefficients for e.g. Faiman, SAPM would not be the right values to use in a model including radiative loss terms, I think. Is there enough data at hand to come up with default coefficients for the new temperature model, or to convert from the generic coefficients?

adriesse, 2022-11-17:
> The point in the technical report is that temperature models fit measurements better when a radiative loss term is included. Is that correct?
> 
Yes.

> The generic coefficients for e.g. Faiman, SAPM would not be the right values to use in a model including radiative loss terms, I think. Is there enough data at hand to come up with default coefficients for the new temperature model, or to convert from the generic coefficients?

You're quite right, those would need to be changed to something (smaller).  I'm working on a data set and demo as well, and will reflect on what values to suggest.  In general I must say that I am trending toward having fewer default values filled in to encourage users to think more about what they're doing.  This might be a situation where no defaults would be appropriate in the function, but with sample values in the Notes.

adriesse, 2022-11-17:
> @adriesse what kind of TV you been watching?

As long as this issue is moving forward: None!

kanderso-nrel, 2022-11-18:
An idea, perhaps a bad one: as an alternative to creating `*_rad` variants of existing models, is it possible and practical to implement the radiative loss term as a standalone generic secondary function kind of like `temperature.prilliman`?

adriesse, 2022-11-18:
Not a bad thought.  I did actually consider some other approaches, like preprocessing/reducing poa_global or using a factory function to make the rad models on demand.  Or something cool with a class again.  But I didn't actually try them out.  

For parameter fitting it's certainly much easier to have a new function like this one, and I don't think I'll do more than one in the near future anyway.

adriesse, 2022-11-20:
> You're quite right, those would need to be changed to something (smaller). I'm working on a data set and demo as well, and will reflect on what values to suggest. In general I must say that I am [...]

The default behaviour with ir_down==None is now same as Faiman, so those u0/u1 default values might still be appropriate.

cwhanse, 2022-11-21:
I like the design @adriesse

adriesse, 2022-11-21:
I could potentially rename the parameters, e.g. u0r and u1r.

cwhanse, 2022-11-21:
I think it's better with the same names as `faiman`. I'm on the fence whether this is an extension of faiman and could be added into that existing function, or, kept separate as a separate model. I can see benefits either way.

kanderso-nrel, 2022-12-12:
@adriesse should we close this now that #1595 is merged, or do you plan to add additional augmented model implementations someday?

adriesse, 2022-12-12:
We can open another issue if/when the demand for additional models becomes overwhelming.

closed 2022-12-12 by adriesse