pvlib/pvlib-python #1601: In appropriate reference for the function "dc_ohmic_losses"

adriesse, 2022-11-25:
**Describe the bug**
The function ```dc_ohmic_losses``` consists of one line:
```
return resistance * current * current
```
which is clearly an elementary electrical calculation.

The pvlib documentation provides a reference to the PVsyst documentation and organizes this function under the PVsyst banner in the documentation.

I think this is a bit strange, and could give the impression that this is something invented by/for PVsyst.

The same could be said about the function ```dc_ohms_from_percent```.

**Desired behavior**
It would be preferable to have basic functions like this without such specific attributions/associations.  A reference, if judged necessary, could be to a simple textbook.  It could be useful to provide the information that PVsyst uses these elementary equations, which could be mentioned in the Notes section perhaps--ideally with information about other software too.

cwhanse, 2022-12-01:
Listing of this function in the Pvsyst group of "PV System Models" documentation is meant to communicate that Pvsyst applies ohmic losses using this method (the "Array ohmic losses page" in the Help document). It wasn't intended to imply that Pvsyst originated the equation (the latter is usually indicated by the name of the function).

The function could certainly be listed in other groupings in the API documentation.

Maybe we should explain what we mean by "PV System Models" - in my mind these are collections of functions that together emulate some named model.

adriesse, 2022-12-01:
Yes, and perhaps we should try to distinguish between models and software.  For me, PVsyst is a software package which implements some generic/basic models/equations, some tweaked standard models, and some original models.

wholmgren, 2022-12-01:
Perhaps we can just change these lines (as the title of the issue suggests) and leave the API documentation questions for other day since that's already in flux in other places:

https://github.com/pvlib/pvlib-python/blob/8e0b724f2514e53b60b3edba2db7c66be87867dd/pvlib/pvsystem.py#L3317-L3321

https://github.com/pvlib/pvlib-python/blob/8e0b724f2514e53b60b3edba2db7c66be87867dd/pvlib/pvsystem.py#L3352-L3356

PR #2229 (2024-09-27): Revise dc_ohmic_loss and dc_ohms_from_percent docstrings
 - [X] Closes #1601
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~~[ ] Tests added~~
 - ~~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link  [... cut]
closed 2024-09-27 by cwhanse