pvlib/pvlib-python #1608: v0.9.4 release plan

kanderso-nrel, 2022-12-09:
Open candidates for inclusion:
- #1558
- #1585
- #1598
- #1544 and #1602
- #1594 and #1595
- #1603 and #1606

Things to defer:

- #1323
- #1345 and #1433
- #1405
- #1443
- #1556
- #1083 and #1080
- #1591
- #1573

@pvlib/pvlib-maintainer please edit as you see fit.  Any objection to choosing the 15th as target release date?

cwhanse, 2022-12-09:
List looks OK to me.

adriesse, 2022-12-20:
> Should we add #1617 and #1618 to the list? They are small and easy to review but are not currently tagged with a milestone.

Only do as much as you can.  I'm trying hard to drift away...

wholmgren, 2022-12-20:
I'm -1 on last minute API changes to well-loved functions. Seems to me that deprecation of `pvwatts` functions and introduction of `pvarray.py` might require more thought than we can put in at the last minute. I'm only now coming up for air after the end of year rush, so apologies for not engaging sooner. I'll bite my tongue if you all want to move ahead with what's already in motion.

kanderso-nrel, 2022-12-20:
@wholmgren I don't mind revisiting any decisions here.  To clarify, are you suggesting (1) reverting #1558 and (2) adding the new functionality in #1602 to `pvsystem.py` instead of creating a new module?

wholmgren, 2022-12-20:
That's one path forward. Another option is reverting #1558 but proceeding with #1602 as is. Another is delaying the release until we have the major API right - maybe that's a day, maybe that's a month.

adriesse, 2022-12-20:
@wholmgren could you be more specific about what you do and don't like?  Top two scenarios for moving forward?

wholmgren, 2022-12-20:
My recommendations are:

1. Quickly engage with Kevin's concern on module name https://github.com/pvlib/pvlib-python/issues/1544#issuecomment-1335696024 but recognizing that it's not a big deal to change this if we need to (I'd do it without deprecation going to 0.10).
2. Merge #1602
3. Do not add more functionality to `pvarray.py` (or whatever name we go with) at this time.
4. Consider if #1558 should be reverted until there is a better home for `pvwattsv5_dc` and `pvwattsv5_losses`. If the answer is "revert", then commit to finding those homes in the next month. If the answer is "keep it", then be comfortable with a second round of deprecations for those functions.
5. Release

kanderso-nrel, 2022-12-20:
I'm okay with keeping `pvarray.py` and possibly renaming it later.  I'm also okay with reverting #1558 and will submit a PR for that shortly.

adriesse, 2022-12-20:
Sounds like a plan!

PR #1620 (2022-12-21): Finalize 0.9.4
 - [x] Closes #1608 
 - [x] Pull request is nearly complete and ready for detailed review.
 - [x] Maintainer: Appropriate GitHub Labels (including `remote-data`) and Milestone are assigned to the Pull Request and linked Issue.

Ready for review.  Speak now or forever hold your peace for this (knock on wood) last release of 2022 :)
closed 2022-12-21 by kanderso-nrel