pvlib/pvlib-python #1621: Develop standard approach for grouping model function and corresponding parameter fitting functions

adriesse, 2022-12-21:
I'm pretty sure I've correctly fixed up all the merge conflicts here, but of course we should expect additional conflicts in any open PRs that touch this part of the docs.  I won't bother folks for a green checkmark before merging since several people have already expressed support.

Let's take up @adriesse's point about where functions should live and be listed in a separate issue (Anton, can you open one?).

_Originally posted by @kanderso-nrel in https://github.com/pvlib/pvlib-python/issues/1598#issuecomment-1360052690_

cwhanse, 2022-12-21:
Is this an issue with organizing the documents, the code, or both?

adriesse, 2022-12-21:
It's worth considering the question for both.  Perhaps more of a discussion item than an issue now, but eventually it could lead to some code changes.