pvlib/pvlib-python #1622: Stop maintaining the `pvlib` anaconda channel?

kanderso-nrel, 2022-12-21:
We provide pvlib distribution files in three places: PyPI, the `conda-forge` anaconda channel, and the `pvlib` anaconda channel.  I can't immediately find a link, but @wholmgren and I have both expressed desire to stop maintaining the `pvlib` anaconda channel.

The main reason to stop maintaining it is that (at least in my experience) it's by far the most manual and fiddly of the three distributions to get right. I haven't been able to get it working on Windows for the last several releases, so I've been doing it on the benchmark server instead, and even that required figuring out some magical incantations.  

There is some reason to think that ditching the `pvlib` channel wouldn't inconvenience most users; with the strange exception of v0.7.2, the `pvlib` channel files have historically been downloaded less than their `conda-forge` analogs:

[image]

<details>
  <summary>Code for plot</summary>

```python
import pandas as pd
import requests
import matplotlib.pyplot as plt

cases = ['pvlib/pvlib', 'conda-forge/pvlib-python', 'conda-forge/pvlib']
versions = ['0.7.0', '0.7.1', '0.7.2',
            '0.8.0', '0.8.1',
            '0.9.0', '0.9.1', '0.9.2', '0.9.3', '0.9.4']

data = {}
for case in cases:
    url = f'https://anaconda.org/{case}/files'
    response = requests.get(url, verify=False)
    df = pd.read_html(response.content)[0]
    df['version_cleaned'] = df['Name'].str.split("-").str[-2]
    downloads = df.groupby('version_cleaned')['Downloads'].sum()
    downloads = downloads.reindex(versions)
    data[case] = downloads

df = pd.DataFrame(data)
df.plot.bar()
plt.ylabel('Total Downloads (all builds)')
```
</details>

The proposition:
- v0.9.4 is the last version to be distributed on the `pvlib` channel
- update the [installation page](https://pvlib-python.readthedocs.io/en/stable/user_guide/installation.html#install-standard-release) to either no longer mention the `pvlib` channel, or say that it's no longer maintained
- include a prominent note in the next release's whatsnew
- look into whether it's possible to mark the `pvlib` channel as obsolete or similar so that users get a warning when installing from it

kanderso-nrel, 2022-12-21:
On the face of it, it's a matter of just changing from `conda install -c pvlib pvlib` to `conda install -c conda-forge pvlib`, and similar for conda environment files.  One subtle difference, perhaps a downside from some perspectives, is that (I think) this will cause not only pvlib itself but also its dependencies to be installed from conda-forge instead of the main anaconda channel.  I've not found that to be an issue in practice but would be interested to get input from others with more experience in the conda world (@wholmgren?).

mikofski, 2022-12-21:
+1 drop pvlib channel and only maintain  conda-forge & pypi

wholmgren, 2022-12-21:
I agree. We might as well keep it up for any users that have scripts or environment files that pull from it.

There used to be two reasons to prefer the pvlib channel in some cases:

1. Until recently (0.9.2?), users needed to specify `pvlib-python` as the package on conda forge, so at least some users preferred using pvlib channel's `pvlib` for compatibility with the pypi package name.
2. Pulling packages from main instead of conda-forge. This no longer strikes me as important given the maturity of conda-forge.

There was a substantial period of time when some versions of pvlib on some platforms were not available on the pvlib channel. I don't remember the details. But I never heard any complaints about the situation during that time, so I'd take that to mean it's unimportant.

Seems like we're all in agreement, so we should consider this issue closed after updating the installation instructions rst page and the release instructions wiki.

PR #1624 (2022-12-22): Scrub instructions regarding `pvlib` conda channel
 - [x] Closes #1622
 - [ ] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub  [... cut]
closed 2022-12-22 by kanderso-nrel