pvlib/pvlib-python #1640: Idea: time-interval-averaged solar position

jarataraj, 2023-01-20:
Many weather model outputs are time interval averages. In order to most accurately work with time-interval-averaged data, it might be useful for pvlib to include a way to get average solar position and/or average daytime solar position over a period of time.  

### Example Use of Average Solar Position:
In the "Estimation of Solar Radiation" section of [this](https://link.springer.com/article/10.1007/s00484-020-01900-5) paper, the authors need to convert the average direct solar irradiance through a horizontal plane (FDIR) (see ["Radiation Quantities in the ECMWF model and MARS"](https://www.ecmwf.int/sites/default/files/elibrary/2015/18490-radiation-quantities-ecmwf-model-and-mars.pdf)) into average direct normal irradiance by dividing FDIR by the average cosine solar zenith angle. They integrate their chosen solar position equation during the daytime subset of a time interval to get the average daytime cosine solar zenith angle. Fig. 2  of the paper maps the effect of solar position averaging (most notable during sunrise and sunset). Their algorithm for the integrated cosine solar zenith angle is implemented in the [thermofeel](https://github.com/ecmwf-projects/thermofeel/blob/720911162ba1c614a20d6fd3c87dea9f554c5fd0/thermofeel/thermofeel.py#L240) python library.

adriesse, 2023-02-02:
You are right that the solar position for hourly average data is not well defined and that pvlib is lacking functions to deal with this, so I think this is a good discussion topic and perhaps even a good reason for a change.  Are you able to get what you want by using the code in thermofeel?  Perhaps you could contribute an example to pvlib that does this?  Or if you think that the functionality should be duplicated in pvlib, would you be in a position to create a PR?

jarataraj, 2023-02-06:
I'd be happy to contribute. Some concerns: 
- some of the math and irradiance concepts discussed in these papers are all new territory for me. I could use some help from someone more familiar with these concepts to help ensure my understanding of the [Hogan and Hirahara paper](https://agupubs.onlinelibrary.wiley.com/doi/full/10.1002/2015GL066868) which originates this method is correct and that the code I produce makes sense
- I am a little unsure of the thermofeel implementation. Why did they use Gaussian numerical integration (with which I am not familiar) instead of the analytical method recommended in the paper (Appendix A, expression A1 in the paper)? Why not apply the curvature correction (expression 2 in the paper)? I have asked the thermofeel developers in [thermofeel issue #31](https://github.com/ecmwf-projects/thermofeel/issues/31)
- Another point of confussion in the paper: what is the difference between $\mu_{0}$ and $\mu_{0m}$?
- Currently, I think I should implement expressions A1 and 2, and get the solar declination angle necessary for expression A1 from other functions in pvlib. This can deliver the average cosine solar zenith and average solar zenith over a period of time. What about azimuth?

Thank you for your help!