pvlib/pvlib-python #1642: pvlib.irradiance.disc not working

ang-endgame, 2023-01-24:
Hi, 

I am receiving the following error when using the pvlib.irradiance.disc function 

My code is: 
`dni = pvlib.irradiance.disc(solar_data['SWGDN'],solpos['zenith'],solar_data['DATETIME'],solar_data['PS'])`

With the error being: 
   1358 """
   1359 Estimate Direct Normal Irradiance from Global Horizontal Irradiance
   1360 using the DISC model.
   (...)
   1429 dirint
   1430 """
   1432 # this is the I0 calculation from the reference
   1433 # SSC uses solar constant = 1367.0 (checked 2018 08 15)
-> 1434 I0 = get_extra_radiation(datetime_or_doy, 1370., 'spencer')
   1436 kt = clearness_index(ghi, solar_zenith, I0, min_cos_zenith=min_cos_zenith,
   1437                      max_clearness_index=1)
   1439 am = atmosphere.get_relative_airmass(solar_zenith, model='kasten1966')

lib\site-packages\pvlib\irradiance.py:98, in get_extra_radiation(datetime_or_doy, solar_constant, method, epoch_year, **kwargs)
     96     RoverR0sqrd = 1 + 0.033 * np.cos(B)
     97 elif method == 'spencer':
---> 98     B = solarposition._calculate_simple_day_angle(to_doy(datetime_or_doy))
     99     RoverR0sqrd = (1.00011 + 0.034221 * np.cos(B) + 0.00128 * np.sin(B) +
    100                    0.000719 * np.cos(2 * B) + 7.7e-05 * np.sin(2 * B))
...
-> 1325         raise integer_op_not_supported(self)
   1326     result = self._time_shift(-other)
   1328 elif isinstance(other, Period):

TypeError: Addition/subtraction of integers and integer-arrays with DatetimeArray is no longer supported.  Instead of adding/subtracting `n`, use `n * obj.freq`

Any help will be greatly appreciated. 
Thanks.

cwhanse, 2023-01-24:
@ang-endgame this is probably an issue with the data in `solar_data`. Can you post your question to the Discussion, with a short sample of the data, or to the pvlib google group? Happy to help there.

closed 2023-01-24 by cwhanse

kurt-rhee, 2025-07-01:
@ang-endgame I'm not sure if you are still working on this problem, but if you are the `datetime_or_doy` parameter must be a pandas `DatetimeIndex`