pvlib/pvlib-python #1644: scipy.misc.derivative is deprecated

kanderso-nrel, 2023-01-24:
scipy [v1.10.0](https://docs.scipy.org/doc/scipy/release.1.10.0.html#deprecated-features) (released Jan 3, 2023) has deprecated `scipy.misc.derivative`, a function used by `pvlib.ivtools.sdm.pvsyst_temperature_coeff`.  The deprecation notice mentions some alternatives (although I'm not confident a new dependency is warranted in our case):

> pvlib/ivtools/sdm.py:1347: DeprecationWarning: scipy.misc.derivative is deprecated in SciPy v1.10.0; and will be completely removed in SciPy v1.12.0. You may consider using findiff: https://github.com/maroba/findiff or numdifftools: https://github.com/pbrod/numdifftools

Noticed in this CI run: https://github.com/pvlib/pvlib-python/actions/runs/3992734328/jobs/6848836375#step:9:130

cwhanse, 2023-01-24:
I noticed that too, in other work. scipy is not going to provide any numeric differentiation tools. If we keep the pvsyst_temperature_coeff function (I vote yes), and are considering a new dependency, I vote against either autograd or its successor, JAX; they are awkward since they wrap numpy. My preference would be to code our own calculation using a finite difference formula.

mikofski, 2023-01-24:
Maybe check [statsmodels](https://www.statsmodels.org/stable/index.html) I seem to recall that it has a simple derivative function, but I can't find it quickly now so maybe I was mistaken?

adriesse, 2023-01-26:
+1 for explicit finite difference calculation

PR #1674 (2023-02-27): Use explicit finite difference instead of scipy.misc.derivative in pvsyst_temperature_coeff

 - [x] Closes #1644
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) [... cut]
closed 2023-02-27 by mikofski