pvlib/pvlib-python #1646: Improve handling of singletons in Perez sky diffuse function

cwhanse, 2023-01-24:
`pvlib.irradiance.perez` returns some odd results when inputs are singletons, and when GHI is 0.

- when used through `pvlib.irradiance.get_total_irradiance` and the inputs are Series of length 1, returned type is a DataFrame with a single index. This is what I expected.
- when used directly as `pvlib.irradiance.perez` and inputs are Series of length 1, a Series of length 2 is returned. This looks like a broadcasting issue.

When irradiance is zero (GHI, DNI and DHI), I expected all irradiance components to be returned as zero. Instead, some components are returned as NaN, or the code errors.
- when used either directly as `pvlib.irradiance.perez` or indirectly via `pvlib.irradiance.get_total_irradiance`, and inputs are floats, division by zero is raised. I expected zero.
- when inputs are Series of length 1 containing zero, `pvlib.irradiance.get_total_irradiance` returns a DataFrame with NaNs in some components, and `pvlib.irradiance.perez` returns a Series (of length 2) with NaN.

**Describe the solution you'd like**
- length 1 inputs return output of length 1, always.
- If GHI, DNI and DHI are all zero (nighttimes in many TMY3 files) all irradiance components are returned as zero.

**Additional context**

This came to light while investigating a [question ](https://groups.google.com/g/pvlib-python/c/pLYPqfdSq9c)posed to the Google group.

cwhanse, 2023-01-24:
Code to illustrate the issue

```
import numpy as np
import pandas as pd
import pvlib

df = pd.DataFrame(data=np.array([[85], [227], [0], [0], [0], [10]]).T,
                  columns=['zenith', 'azimuth', 'dni', 'ghi', 'dhi', 'airmass'])

# use through get_total_irradiance
perez_wrapped = pvlib.irradiance.get_total_irradiance(
    surface_tilt=90, surface_azimuth=90,
    solar_zenith=df['zenith'], solar_azimuth=df['azimuth'],
    dni=df['dni'], ghi=df['ghi'], dhi=df['dhi'],
    dni_extra=1413, albedo=0., model='perez')

print(perez_wrapped)

perez_natural = pvlib.irradiance.perez(
    surface_tilt=90, surface_azimuth=90,
    dhi=df['dhi'], dni=df['dni'], dni_extra=1413,
    solar_zenith=df['zenith'], solar_azimuth=df['azimuth'],
    airmass=10.)

print(perez_natural)

perez_float = pvlib.irradiance.get_total_irradiance(
    surface_tilt=90, surface_azimuth=90, solar_zenith=85, solar_azimuth=227,
    dni=0., ghi=0., dhi=0., dni_extra=1413, albedo=0., model='perez')

perez_float = pvlib.irradiance.perez(
    surface_tilt=90, surface_azimuth=90, solar_zenith=85, solar_azimuth=227,
    dni=0., dhi=0., dni_extra=1413, airmass=10.)

```

wholmgren, 2023-01-24:
I agree the divide by 0 exception does not conform to scientific python expectations. The `perez` function is already firmly in pandas/numpy territory, so I don't object to the easy solution of adding `dhi = np.asarray(dhi)  # GH1646` before the calculation of `eps`.

Returning 0 in that case is more complicated because we want to allow for scalar inputs and to preserve nans. The most straightforward might be to also wrap `dni` in `asarray` and then after the `eps` calculation add a line 

`eps[np.isnan(eps) & ~(np.isnan(dni) | np.isnan(dhi))] = 0  # 0/0->0 GH1646`

Alternatively we can use `np.true_divide` with `where` and `out` arguments. See `pvsystem.sapm` for one example.

Not sure about the broadcasting problem but the solution above might also fix it.