pvlib/pvlib-python #1647: get_psm3() reorders the dataframe columns randomly

Arjannikov, 2023-01-24:
Every time I call get_psm3(), the returned data frame column order is different for the same query parameters.

**CODE EXAMPLE**

for count in range(3):
     df, metadata = pvlib.iotools.get_psm3(
                latitude=latitude, longitude=longitude,
                api_key=NREL_API_KEY,
                email=MY_EMAIL_ADDRESS
                names=2019,
                interval=60,
                map_variables=True,
                leap_day=True)
     print(df.columns)

**OUTPUT**

Index(['Year', 'Month', 'Day', 'Hour', 'Minute', 'pressure', 'Dew Point', 'dni', 'ghi', 'dhi', 'albedo', 'temp_air', 'wind_speed', 'wind_direction'], dtype='object')

Index(['Year', 'Month', 'Day', 'Hour', 'Minute', 'Dew Point', 'pressure', 'wind_speed', 'dni', 'dhi', 'albedo', 'wind_direction', 'ghi', 'temp_air'], dtype='object')

Index(['Year', 'Month', 'Day', 'Hour', 'Minute', 'dhi', 'Dew Point', 'pressure', 'ghi', 'temp_air', 'wind_direction', 'dni', 'albedo', 'wind_speed'], dtype='object')

**Versions:**

 - ``pvlib.__version__``: '0.9.4'
 - ``pandas.__version__``: 1.5.3
 - python: sys.version_info(major=3, minor=9, micro=13, releaselevel='final', serial=0)

cwhanse, 2023-01-24:
Strange. I don't reproduce this behavior with Window/python 3.9.7, pandas 1.5.3, pvlib 0.9.4. What OS are you on?

kanderso-nrel, 2023-01-24:
I can't reproduce different column orderings in each loop iteration in a single python session, but I can get different orderings by killing and starting a fresh python session.  I am not sure why that would have any effect.  

I checked the raw data returned from the PSM3 API call and it seems that the random ordering is on the API's part, not ours -- we just pass through the ordering we get (at least with `map_variables=False`, I didn't check with `True`).  I thought I understood that the API had a layer where repeated calls with identical parameters would get a cached copy of the corresponding data.  I can't square that with these observations.

Worth pointing out that a big PSM3 data update just took place; seems plausible that this is connected.

kanderso-nrel, 2023-01-24:
Ah ha:

https://github.com/pvlib/pvlib-python/blob/b11c9e2e55db71e7cac611717a027efb3761e401/pvlib/iotools/psm3.py#L161-L166

I suspect the `set` operation is to blame here: our API request includes the columns in random order and the API returns data to match.

PR #1648 (2023-02-02): Fix reverse mapping issue and undesirable column reordering in get_psm3

 - [x] Closes #1629 and closes #1647
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for al [... cut]
closed 2023-02-02 by kanderso-nrel