pvlib/pvlib-python #1649: Accelerate singlediode._lambertw

cwhanse, 2023-01-25:
I've found a way to replace the calculation of the maximum power point in `singlediode._lambertw` with a more analytic approach. The speedup is not quite 10x, and would make `pvsystem.singlediode(...method='lambertw')` and `pvsystem.singlediode(...method='newton')` comparable.

I'm not sure if the maths are new, they might be. In any case, it would not be difficult to document on pvpmc.sandia.gov, if not in a conference paper.

We could remove the golden mean search, or retain it. I lean towards retaining it. The new way would use `scipy.optimize.newton` which should converge with quite high confidence, but, not guaranteed for some pathological case. golden mean search would always converge.

[gist](https://gist.github.com/cwhanse/25b91e01d7de594399eb97fb010499ac) with new method.

adriesse, 2023-02-01:
I looked at the gist but was unable to get the gist of it.  :(  But if it works, why not?

cwhanse, 2023-02-01:
@adriesse fair point, I will document this work before proposing changes to pvlib code.

closed 2026-03-13 by cwhanse