pvlib/pvlib-python #1659: modelchain.results.ac.plot(figsize=(16,9)) doesnot yield any power output

arijit91c, 2023-02-13:
I am working for the first time with pvlib-python and written a simple script to generate ac power output. Just a clear sky weather model, pvsyst freestanding temperature model, aoi model = ashrae and spectral loss model = no loss. But when I am running the modelchain and plotting the output, I am getting  a flat line. Am I missing something here? Attached is the .txt  file for reference.

[pvlib_001.txt](https://github.com/pvlib/pvlib-python/files/10722827/pvlib_001.txt)

[image]

pvlib version: 0.9.4

kanderso-nrel, 2023-02-13:
Hi @arijit91c, the problem is that your `PVSystem` is defaulting to only a single PV module, which does not supply nearly enough to turn on the large 2.5 MW inverter.  You can see this by running `print(system)` and seeing that it shows `1` for both `strings` and `modules_per_string`. Try setting `modules_per_string=8, strings_per_inverter=1000` (or whatever reflects the system you are simulating) in the `PVSystem` constructor for a more realistic simulation.  For more information on the available parameters, see https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.PVSystem.html

Also, we use GitHub Issues just for tracking bugs and feature requests in pvlib; please direct usage questions to [Discussions](https://github.com/pvlib/pvlib-python/discussions) or the google group :)

arijit91c, 2023-02-13:
Hi @kanderso-nrel, thanks for the heads up. This works as expected.

closed 2023-02-13 by arijit91c