pvlib/pvlib-python #1663: Urgent: Update project size

henrykironde, 2023-02-14:
Currently, you have "Long (350hr) or Medium (175hr) depending on student expertise,"  but Google needs us to specify either 350 or 175 for logistical purposes.

https://github.com/pvlib/pvlib-python/wiki/GSoC-2023-Projects

AdamRJensen, 2023-02-14:
Hi @henrykironde 

To me, it seems that all of our project ideas state the length (175 or 350 hours), so I'm not quite sure what it is that you are asking?

[image]

henrykironde, 2023-02-14:
Choose one.

AdamRJensen, 2023-02-14:
Thanks for clarifying @henrykironde. I have gone ahead and updated the project ideas list.

@kanderso-nrel Please check if you agree with my changes.

closed 2023-02-14 by AdamRJensen