pvlib/pvlib-python #1669: Implement all-sky spectral irradiance model(s)

kanderso-nrel, 2023-02-20:
**Is your feature request related to a problem? Please describe.**
pvlib's spectral irradiance modeling is currently limited to `pvlib.spectrum.spectrl2`, a clear-sky model.  As cloudy-sky spectral irradiance takes a different shape from that of clear-sky irradiance, simulating cloudy-sky spectral irradiance is currently a gap in pvlib's capabilities.

**Describe the solution you'd like**
[SEDES2](https://doi.org/10.1016/0927-0248(92)90083-2) and [TMYSPEC](https://doi.org/10.1016/j.solener.2012.06.014) are extensions/add-ons to SPECTRL2 that use broadband irradiance to calculate modifiers that transform clear-sky spectral irradiance to cloudy-sky spectral irradiance.  With a python-native SPECTRL2 already implemented, I think it would not take much code to implement one of these as well.  However, although the models aren't very complex, it might be difficult to find all the details (coefficient tables and such) necessary for implementation.

On that note @shirubana has kindly sent me a C# implementation of SEDES2 from Bill Marion (used in [NREL/TP-520-47277](https://www.nrel.gov/docs/fy10osti/47277.pdf)) as well as a python translation of her own creation.  This code claims to be a descendant of Fortran code dated Aug 1 1992 developed by Centre for Solar Energy and Hydrogen Research (ZSW) in Germany by Stefan Nann and Angelika Bakenfelder.

**Describe alternatives you've considered**
N/A

**Additional context**
I think @adriesse has been involved with some studies that used SEDES2.

See also #857 (SMARTS) and #71 (PyRTM).

shirubana, 2023-02-20:
tx for starting this!

I seem to recall talking with Jean Schulte for releasing this and it was possible, but we should make sure we establish the proper copyright/legal path of emails for this...

adriesse, 2023-02-20:
Yes, I have worked with all of these and I'm pretty sure you could find all the information that is needed to re-implement them.  There are historically significant and it would be an interesting activity for sure.

mikofski, 2025-12-09:
ICYMI @jdnewmil