pvlib/pvlib-python #1679: Multiple inverters for one system

HannanSolo, 2023-03-01:
Is it possible to include multiple inverters in a system or model chain?

**Describe the solution you'd like**
It would be ideal to include an inverter parameter in the array object that overrides the pvsystem parameter.

I have looked at #1132 and https://groups.google.com/g/pvlib-python/c/qLSN-3h0N6Y and seems it's just an unimplemented feature.

The current way we are able to computer dc power is just multiplying the inverter parameters by the array amount.

```
system = pvlib.pvsystem.PVSystem(
    arrays=arrays,
    inverter_parameters=inverter * num_array
)
```

Which feels very hacky.

cwhanse, 2023-03-01:
The challenge here is that each inverter should have its own set of arrays.

cwhanse, 2023-03-01:
If this works (a ModelChain runs without error), that's not intentional. I haven't tried it.

```
system = pvlib.pvsystem.PVSystem(
    arrays=arrays,
    inverter_parameters=inverter * num_array
)
```

HannanSolo, 2023-03-01:
Hey Cliff, thanks for the very active communication.

It does not work with a ModelChain, only if manually computing the output: see below.

```
# Sum over multiple dataframes.
def sum_over_dataframes(candidates):
    return reduce(lambda a, b: a.add(b, fill_value=0), candidates)

dc_power = system.scale_voltage_current_power(mpp)
sum_dc_power = sum_over_dataframes(dc_power)
sum_dc_power
#calculate ac power.
ac_powers = []
for dc_power_item in dc_power:
    ac_power = pvlib.inverter.sandia(v_dc=dc_power_item.v_mp, p_dc=dc_power_item.p_mp, inverter=inverter)
    ac_powers.append(ac_power)
sum_ac_power = sum_over_dataframes(ac_powers)
```
This then returns a pretty similar output of our array when comparing it to pvSyst.

But feel very hacky and wondered if there is a better way.

HannanSolo, 2023-03-01:
Maybe it's worth structuring Inverters as its own class that contains arrays attached to the inverter and have PVSystems as a collection of inverter/array pairs.

Or have the option to include an optional inverter as a parameter of Array.

Not sure how possible it would be to add it, but it's what comes to mind first.