pvlib/pvlib-python #1681: v0.9.5 release planning

kanderso-nrel, 2023-03-02:
Keeping to the quarterly release schedule means releasing 0.9.5 sometime in mid March.  I arbitrarily suggest March 15 for the target release date.

Looking through the [v0.9.5 milestone](https://github.com/pvlib/pvlib-python/milestone/39), these PRs seem like they are close to being ready to merge and are thus candidates for inclusion:

- [ ] #1179
- [ ] #1573
- [ ] #1623
- [ ] #1628
- [ ] #1655
- [ ] #1658
- [ ] #1666
- [ ] #1672

@pvlib/pvlib-core please review any PRs that interest you.  I volunteer to solicit reviewers for PRs that don't attract attention on their own over the next couple weeks.

cwhanse, 2023-03-02:
I will submit to close #1586

adriesse, 2023-03-02:
Based on the latest comments, I would not wait for #1658.

kanderso-nrel, 2023-03-16:
Assuming nobody objects, I'll plan to merge #1183 and #1682 tomorrow (March 17) and proceed with making the release.  The other PRs in the list above are not ready yet, I think.

cwhanse, 2023-03-16:
I'm working on #1666 but computing the known solution for the test case is tedious. I wouldn't wait.

adriesse, 2023-03-16:
I'd like to have #1695 included, it there are no objections.

mikofski, 2023-03-16:
If we wait on #1666, then I'd like to push another release as soon as possible after this one. A hot fix.

mikofski, 2023-03-16:
> Assuming nobody objects, I'll plan to merge #1183 tomorrow (March 17) ...

@kanderso-nrel oops sorry, I merged it already 🤷

wholmgren, 2023-03-17:
What about #1687 in 0.9.5? It's currently marked for 0.10, but @mikofski and the contributor are resolving merge conflicts in 0.9.5, so there's some misaligned expectations. I am -1 on 0.9.5 given the lack of consensus on if it's a good idea and, from a quick look, a need to improve consistency with pvlib style (variable names, positional vs keyword args).

cwhanse, 2023-03-17:
I marked #1687 for v0.10 because the comments indicate a lack of consensus.

mikofski, 2023-03-17:
I agree, let's push #1687 to v0.10. I'll communicate with contributor

PR #1699 (2023-03-18): Finalize 0.9.5
 - [x] Closes #1681
 - [x] Pull request is nearly complete and ready for detailed review.
 - [x] Maintainer: Appropriate GitHub Labels (including `remote-data`) and Milestone are assigned to the Pull Request and linked Issue.

Ready for review
closed 2023-03-18 by kanderso-nrel