pvlib/pvlib-python #1683: Sandia Inverter Fit Algorithm gives Efficiencies > 1

kurt-rhee, 2023-03-05:
**Describe the bug**
It is possible that I am using the fit parameters incorrectly.  Attempting to fit the Sandia inverter model to parameters from the SMA Sunny Central 4400 and then predicting efficiency gives efficiency values >1 at low power inputs.  

[image]
Figure 1:  Sandia Inverter Efficiencies (red curve), PVsyst inverter efficiencies (blue curve)

**To Reproduce**
Steps to reproduce the behavior:

1. Use the following parameters from PVSyst as input parameters for the pvlib.inverter.fit_sandia method

```
dc_power = pd.DataFrame({
    'Vmin': [225000, 447000, 891000, 1336000, 2227000, 3345000, 4468000],
    'Vnom': [227000, 449000, 893000, 1338000, 2230000, 3350000, 4477000],
    'Vmax': [228000, 449000, 893000, 1339000, 2231000, 3351000, 4479000]
})

efficiency = pd.DataFrame({
    'Vmin': [0.9768, 0.9847, 0.9879, 0.9883, 0.9879, 0.9866, 0.9847],
    'Vnom': [0.9677, 0.9801, 0.9853, 0.9863, 0.9864, 0.9852, 0.9827],
    'Vmax': [0.9655, 0.9795, 0.9849, 0.9857, 0.9859, 0.9847, 0.9823]
})

ac_power = efficiency * dc_power

dc_voltage = pd.DataFrame({
    'Vmin': [962, 962, 962, 962, 962, 962, 962],
    'Vnom': [1144, 1144, 1144, 1144, 1144, 1144, 1144],
    'Vmax': [1193, 1193, 1193, 1193, 1193, 1193, 1193]
})
dc_voltage_level = pd.DataFrame({
    'Vmin': ['Vmin', 'Vmin', 'Vmin', 'Vmin', 'Vmin', 'Vmin', 'Vmin'],
    'Vnom': ['Vnom', 'Vnom', 'Vnom', 'Vnom', 'Vnom', 'Vnom', 'Vnom'],
    'Vmax': ['Vmax', 'Vmax', 'Vmax', 'Vmax', 'Vmax', 'Vmax', 'Vmax']
})
p_ac_0 = 4400000
p_nt = 370

ac_power = ac_power.values.flatten()
dc_power = dc_power.values.flatten()
dc_voltage = dc_voltage.values.flatten()
dc_voltage_level = dc_voltage_level.values.flatten()
```

2.  Use the values to fit the sandia inverter model

```
fit = pvlib.inverter.fit_sandia(
    ac_power=ac_power,
    dc_power=dc_power,
    dc_voltage=dc_voltage,
    dc_voltage_level=dc_voltage_level,
    p_ac_0=p_ac_0,
    p_nt=p_nt
)
```

3. Predict inverter output 
 [pvsyst_inverter.CSV](https://github.com/pvlib/pvlib-python/files/10893125/pvsyst_inverter.CSV)

```
df = pd.read_csv('pvsyst_inverter.csv', na_values=['#DIV/0!'])
df['Efficiency'] = df['EOutInv'] / df['EArray']
df['EArray'] = df['EArray'] * 1000

df['EOutInv_Sandia'] = pvlib.inverter.sandia(
    v_dc=df['UArray'],
    p_dc=df['EArray'],
    inverter=fit
)
df['Efficiency_Sandia'] = df['EOutInv_Sandia'] / df['EArray']
```

**Expected behavior**
I would expect values to always be below 1

**Screenshots**
Included above

**Versions:**
 - ``pvlib.__version__``: 0.9.4
 - ``pandas.__version__``: 1.5.3
 - python: 3.9

**Additional context**
Add any other context about the problem here.

kanderso-nrel, 2023-03-06:
The rows in `pvsyst_inverter.csv` corresponding to `EOutInv_Sandia > EArray` have array voltage far below the fitted `Vmin`.  The PVsyst inverter doesn't operate during those conditions and outputs its `Pnt`:

```python
In [48]: df.loc[df['Efficiency_Sandia'] > 1, ['UArray', 'EArray', 'EOutInv']].describe()
Out[48]: 
           UArray       EArray  EOutInv
count   21.000000    21.000000    21.00
mean   623.288571  7100.261905    -0.37
std      0.720425    19.037397     0.00
min    622.220000  7073.300000    -0.37
25%    622.910000  7090.300000    -0.37
50%    623.140000  7096.000000    -0.37
75%    623.790000  7112.900000    -0.37
max    625.160000  7152.200000    -0.37
```

But `pvlib.inverter.sandia` doesn't do that check (it can't, since it doesn't know the inverter's operating voltage range).  The docstring says: 

> `power_ac` is not adjusted for maximum power point tracking (MPPT) voltage windows or maximum current limits of the inverter.

So I think this is expected behavior, albeit not the most user-friendly.

kurt-rhee, 2023-03-06:
Hey @kanderso-nrel, that was a great explanation.  Thank you, I will close this one out.

closed 2023-03-06 by kurt-rhee

kanderso-nrel, 2023-03-06:
Thanks @kurt-rhee for the reproducible report!  Sorry, I didn't necessarily mean that there isn't anything to improve here, I just wanted to point out what the problem is, and that the limitation is documented.  Brainstorming options:

The CEC inverter database does include `Mppt_low`, `Mppt_high`, and `Idcmax`, so it might be reasonable to add those types of checks to `inverter.sandia`.  Users with parameters from `fit_sandia` could be required to add those keys themselves.

Another option, in lieu of implementing those checks, is to just add a constraint like `power_ac = np.clip(power_ac, a_max=p_dc)`.  Or even just emit a warning if `any(power_ac > p_dc)`.

cwhanse, 2023-03-06:
My thoughts: the published model assumes that the inverter stays in the voltage range for its MPPT. I think it's OK to extend the model beyond the publication as long as we document in the function's documentation.

I learned last week that the MPPT voltages are not always constants. The values on the datasheets are those at nominal conditions. More motivation to develop and publish a better inverter model.