pvlib/pvlib-python #1694: Diffuse Irradiance Optimization for Single Axis Trackers

kurt-rhee, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
Some tracker manufacturers have implemented a diffuse irradiance optimization algorithm into their controls.  The algorithm determines which angle the tracker should track to in order to maximize plane of array irradiance given the current irradiance conditions in terms of DHI and DNI.  

**Describe the solution you'd like**
I would like to model a generic diffuse irradiance optimization algorithm in pvlib.  

**Describe alternatives you've considered**
Instead of implementing this feature inside of the library, it could be posted as a user guide.  

**Additional context**
PlantPredict is about to release a new 2 parameter diffuse irradiance optimization algorithm in our next major version release.  The two parameters are meant to reduce overestimation from irradiance optimization algorithms which are too ideal.  I used pvlib in order to test the algorithm before having it implemented in C# and would be happy to contribute the code to pvlib (if there is any interest from the maintainers/community of course).  The is no paper yet unfortunately.  My hope is to publish at next year's PVSC. 

There is some available documentation here:

https://terabase.atlassian.net/servicedesk/customer/portal/3/article/1260060675

High level notes:
- For loop through all tracking angles from minimum rotation angle to maximum rotation angle with a step of 1 degree
- Choose whichever angle gave the best plane of array irradiance
- If a tracker rotation speed or a hesitation factor are != 0
- Calculate what percentage of the timestep the tracker is in motion, what percentage of the timestep the tracker hesitates before beginning its movement towards the ideal tracker angle
- Create a corrected angle which is the weighted average angle between the ideal tracker angle, the in transit tracker angle and the hesitation tracker angle.
-  If tracker rotation speed and hesitation factor are == 0
-  Diffuse optimization algorithm is ideal
- Return the corrected angle

echedey-ls, 2024-01-27:
I'm going to address this issue if no objections are made.

The _high level notes_ section is the methodology used in order to define the optimization algorithm, right? I still have to read a few papers before a PR. But from the web page, a PVLIB implementation seams feasible.

kurt-rhee, 2024-01-27:
Hey @echedey-ls that is correct, if it is helpful, I can provide the C# code here as a comment.

echedey-ls, 2024-01-27:
Glad to hear that @kurt-rhee !

I will have to write the tests and from a quick glance at the papers, I don't have a clue on where to start. So maybe checking against that could be an option? Maybe there's an easier path. I don't really see another reason to share them, but who knows.

If it's not a big deal for you then, please provide it. Anyways, thanks for the fast response!

kurt-rhee, 2024-01-27:
Of course!  I would like to see if the maintainers are interested in including this feature in pvlib though before starting.  Since there is no paper that specifies the algorithm, it may fall beneath the bar for inclusion.

mikofski, 2024-01-27:
@kurt-rhee  there is that Terabase link right in Atlassian, any way to upload that to zenodo?

kurt-rhee, 2024-01-27:
@mikofski absolutely, I'll shoot for having a copy up there in one week

kandersolar, 2024-01-27:
I think there is room for these kinds of models in pvlib.  @kurt-rhee is a paper for this model in the works?  I need to read that docs page in more detail, but on first glance I'm not sure its content is detailed enough to act as a reference for a pvlib implementation.

Also FYI I notice a minor inconsistency on that webpage: it says that $\mu$ and $\eta$ range from 0-100, but the equation for $\theta_c$ treats them as quantities from 0-1.

kurt-rhee, 2024-01-28:
@kandersolar thanks for pointing out that error in the documentation, I have corrected it.

As for the paper, I haven't started it, but I wouldn't mind writing one.  I only have experience writing papers for the PVSC conferences, do you have any suggestions of where else I could submit it?  We just passed the submission window for this year's PVSC.

kurt-rhee, 2024-01-29:
@mikofski 

I wasn't able to get into Zenodo unfortunately, looks like they are having some issues with their login right now.  I tried on firefox and chrome. Open issue on github (https://github.com/zenodo/zenodo/issues/2176)

mikofski, 2024-01-29:
PVPMC would be a good reference - wish we could also share some validation data, maybe check with @williamhobbs or @abhisheksparikh

williamhobbs, 2024-01-29:
I'd be glad to chat about possible validation data. I think folks can find my work email via LinkedIn, or personal email in the pvlib google group. 

Outside of NREL arrays, the NIST demo site, etc., some relatively new _public_ datasets options are:
 - DOE Solar Data Bounty Prize sites: see https://openei.org/wiki/PVDAQ, https://doi.org/10.25984/1846021 (and the upcoming DOE/NREL webinar)
 - EPRI SSRC: data here https://github.com/epri-dev/PV-Subhourly-Clipping, briefly discussed here https://github.com/pvlib/pvlib-python/discussions/1889#discussion-5743633. An EPRI report describing the dataset in more detail is pending, but I'd be glad to chat offline to fill in gaps.

adriesse, 2024-01-29:
> PVPMC would be a good reference

This is an open, unanswered question: #1898

kurt-rhee, 2024-01-30:
@mikofski I agree that some validation would be great.  I won't be given time this year to work on a validation study, but if anybody here now or in the future is interested in studying it I can provide the code in python, C#, or a research license to PlantPredict

williamhobbs, 2026-06-26:
The links that wrap over to the next line seem broken for me. I realized ref 11 is the same. Viewing the pdf in Firefox and Edge. Not that it matters too much at this stage...

[image]