pvlib/pvlib-python #1697: poa_horizontal_ratio is untested

mikofski, 2023-03-13:
**Describe the bug**
Currently `pvlib.irradiance.poa_horizontal_ratio` is untested according to codecov

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://app.codecov.io/gh/pvlib/pvlib-python/blob/main/pvlib/irradiance.py#L259

**Expected behavior**
100% test coverage

**Screenshots**
[image]

**Versions:**
 - ``pvlib.__version__``: 0.9.5-dirty
 - ``pandas.__version__``: ?
 - python: ?

**Additional context**
This causes codecov to fail for any edits to irradiance.py that reduce the number of lines because then code coverage goes down.

wholmgren, 2023-03-13:
Apparently I added this function 9 years ago during a large refactor, perhaps in an aborted effort at DRY code, but never used it. Tests were hardly a thing at that time. In any case, I suggest we delete it.

Lakshyadevelops, 2023-03-19:
@wholmgren Function pvlib.irradiance.poa_horizontal_ratio is being used in example irradiance.ipynb
also what is the following checking for?
```
try:
    ratio.name = 'poa_ratio'
except AttributeError:
    pass
```
 name attribute of a float?same try-except is for many other functions in irradiance module

matsuobasho, 2023-11-10:
Hi, I would like to tackle this issue.  It seems like there is more to it aside from just deleting the `poa_horizontal_ratio` function.  If I delete that function, then do I do anything with the `docs/tutorials/irradiance.ipynb` notebook that uses the function?

Thanks.