pvlib/pvlib-python #1721: Broken links

langestefan, 2023-05-02:
**Describe the bug**
Broken links in PVlib docs. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [pvsystem_tutport](https://pvlib-python-dacoex.readthedocs.io/en/pvsystem_tutport/usage.html), or [pvlib tutorials](https://nbviewer.ipython.org/github/pvlib/pvlib-python/tree/main/docs/tutorials/)
2. On tutport all the examples get a 404, nbviewer doesn't load either

kandersolar, 2023-05-03:
Thanks @langestefan!  The first one is a six-year-old unofficial fork of the pvlib documentation (see how the url is `pvlib-python-dacoex.readthedocs` instead of just `pvlib-python.readthedocs`).  We can't fix that one since it's not our webpage.  

We can fix the second one though.  Looks like `docs/sphinx/source/index.rst` has two instances of the outdated `nbviewer.ipython.org` that need to be updated.

langestefan, 2023-05-03:
@kandersolar I see, thanks. Not sure how I found that first link, I think it was somewhere at the top of google search results? Anyway good to know it's outdated, I will refer to the second link in the future. You can close the issue now if you want

kandersolar, 2023-05-03:
Let's keep it open until our outdated nbviewer links are fixed :)

Fixing those would be an easy first PR if anyone is interested.

kshitiz305, 2023-05-04:
Hi @kandersolar I would like to work on this could you help me with the link that needs to be updated.

kandersolar, 2023-05-04:
Hello @kshitiz305, the links in question are in this file: https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/index.rst?plain=1

There are two URLs for `nbviewer.ipython.org/...`.  I think they just need to be updated to `nbviewer.org/...`.

PR #1726 (2023-05-04): Update outdated nbviewer links
Fixes : #1721
:issue:#1721

 - [x] Closes #1721 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [x] [... cut]
commit 06594f96 (2023-05-04): Update outdated nbviewer links (#1726)

* Broken links

Fixes : #1721

* move whatsnew entry to Documentation section

* add 0.9.6 whatsnew to whatsnew index

---------
closed 2023-05-04 by kandersolar