pvlib/pvlib-python #1722: stickler-ci is shutting down

kandersolar, 2023-05-03:
It seems the stickler-ci service is shutting down: https://stickler-ci.com/shutting-down

That website seems to suggest that it should still be working for the moment, but @AdamRJensen noticed that it's not reporting a status to #1395, so it might be inactive already for us.

Thinking about how we might replace it: instead of using an external service, [pvanalytics](https://github.com/pvlib/pvanalytics/blob/main/.github/workflows/lint-and-test.yml#L63) and [twoaxistracking](https://github.com/pvlib/twoaxistracking/blob/main/.github/workflows/lint.yml#L9) both DIY with a custom flake8 action.  This approach doesn't comment in-line like stickler does, which is a downside especially for new contributors.  Probably there are nicer pre-built linter services we could switch to.

mikofski, 2023-05-03:
[Pycodestyle](https://pypi.org/project/pycodestyle/) returns line item results we could pipe that to an artifact. No preference between this and flake8

i also notice there are many GitHub actions for python code style checking. IDK if any will offer the same level of feedback, but maybe?
* this action created stickler like GitHub annotations from flake8 output: https://github.com/marketplace/actions/flake8-annotator
* here’s a long list of flake 8 actions including the annotation one I just mentioned, some that are defunct, & some that are more popular than others: https://github.com/marketplace?type=actions&query=Flake8
* Here’s a list of a few pycodestyle actions: https://github.com/marketplace?type=actions&query=Pycodestyle
* Here’s one for pep8: https://github.com/marketplace?type=actions&query=pep8+
* Here’s another random python code style checker: https://github.com/marketplace?type=actions&query=python+code+style+checker+

kandersolar, 2023-05-04:
> it's not reporting a status to https://github.com/pvlib/pvlib-python/pull/1395, so it might be inactive already for us.

Well I see it's running on other PRs now.  Not sure why it's not running on #1395.  Anyway I guess we don't have to replace it immediately.

adriesse, 2023-05-05:
> It seems the stickler-ci service is shutting down

Yay, no more annoying stickler error messages!!  :)

wholmgren, 2023-06-27:
I'm ok with the custom action, but my preference is to embrace the dark side and go [black](https://black.readthedocs.io/en/stable/index.html). I don't expect consensus on this and I'm not going to push it today. I just want to use this opportunity to plant a seed in your minds :) See details below for our new AI overlord's vision of a world in black

<details>

(Verse 1)
Imagine there's no clutter, no messy code to see,
No tangled lines to fix, just pure simplicity.
Imagine all the spaces, just erased with a tap,
You may say I'm a dreamer, but I'm not the only one.

(Chorus)
I hope someday you'll join us,
And the world will be bug-free,
With the black formatter, it's easy,
To code in harmony.

(Verse 2)
Imagine there's no tabs, only indents in our code,
No more endless debates, over style choices bestowed.
Imagine all the programmers, working hand in hand,
You may say I'm a dreamer, but I'm not the only one.

(Chorus)
I hope someday you'll join us,
And the world will be bug-free,
With the black formatter, it's easy,
To code in harmony.

(Bridge)
You may say I'm obsessed, with formatting rules so strict,
But I believe in consistency, it's a programming pick.
I hope someday you'll see, the beauty that it brings,
A world of clean code, where simplicity sings.

(Verse 3)
Imagine no more wasted time, fixing formatting woes,
No more endless scrolling, to find what we suppose.
Imagine all the coders, focusing on what truly matters,
You may say I'm a dreamer, but I'm not the only one.

(Chorus)
I hope someday you'll join us,
And the world will be bug-free,
With the black formatter, it's easy,
To code in harmony.

(Outro)
Imagine a world of code, where order reigns supreme,
No more tangled spaghetti, just a programmer's dream.
You may say I'm a dreamer, but I'm not the only one,
With the black formatter, we can make it all come undone.

</details>

AdamRJensen, 2023-07-01:
> I'm ok with the custom action, but my preference is to embrace the dark side and go [black](https://black.readthedocs.io/en/stable/index.html). I don't expect consensus on this and I'm not going to [...]

There are some things that Black doesn't do it seems: e.g., flagging unused variables or imports

wholmgren, 2023-07-06:
Good point Adam. We can use one or more of black, ruff, flake8, isort, pydocstyle, etc. [metpy's multi-tool linting workflow](https://github.com/Unidata/MetPy/blob/main/.github/workflows/linting.yml) looks interesting in part because they're piping the output of the linters into [reviewdog](https://github.com/reviewdog/reviewdog). https://github.com/Unidata/MetPy/pull/2967 suggests that their goal is to consolidate most linting into ruff as it gains more features.

mikofski, 2023-07-08:
I’m curious if anyone investigated just using an off the shelf marketplace action like the ones I linked in [this comment](https://github.com/pvlib/pvlib-python/issues/1722#issuecomment-1533360173) before writing custom code in #1786 that we’ll have to maintain?

reepoi, 2023-07-12:
I looked at the flake8 options in the actions marketplace and felt they don't contribute much toward writing a CI workflow that only lints changed lines. There appear to be two kinds of flake8 actions available on the marketplace. The first kind installs and runs flake8 using any custom arguments we provide, and then captures flake8's output to create annotations on the pull request. The second kind only sets up the tool that captures flake8's output for annotations and requires us to install and run flake8. To use these actions, we need to write a workflow file in this repository that calls them, and it is only a few more lines of code to reimplement their functionality ourselves.

The first kind of action does not support linting changed lines only. The easiest way to lint changed lines is to pipe the output of `git diff` to `flake8 --diff`. If the action calls flake8 for us, I don't know how to pipe the `git diff` output to it. Next, the `--diff` option was removed in flake8 v6, so we would need to be able to control the version of flake8 that the action uses.

The second kind of action does so little that reimplementing its functionality is more maintainable. All the implementation details are in one place. The implementation is writing a JSON file that defines how flake8's output should be mapped to annotations and then registering this JSON file with the CI workflow.

closed 2023-07-31 by kandersolar