pvlib/pvlib-python #1724: `pvlib.iotools.get_pvgis_hourly`'s `surface_azimuth` parameter doesn't use pvlib's azimuth convention

kandersolar, 2023-05-03:
Nearly everything in pvlib represents azimuth angles as values in [0, 360) clockwise from north, except `pvlib.iotools.get_pvgis_hourly`:

https://github.com/pvlib/pvlib-python/blob/3def7e3375002ee3a5492b7bc609d3fb63a8edb1/pvlib/iotools/pvgis.py#L79-L81

This inconsistency is a shame.  However, I don't see any way to switch it to pvlib's convention without a hard break, which is also a shame.  I wonder how others view the cost/benefit analysis here.

See also https://github.com/pvlib/pvlib-python/pull/1395#discussion_r1181853794

cwhanse, 2023-05-03:
Since `get_pvgis_hourly` is only using `surface_azimuth` to pass a value to a pvgis keyword `aspect`, we could add `aspect` with the South=0 convention and deprecate `surface_azimuth` as is. Then later, we can deprecate `aspect` and add `surface_azimuth` back with its pvlib meaning. Now that I write that out, seems better just to make the breaking change.

closed 2023-05-25 by kandersolar

tongpu, 2023-08-22:
This issue discusses that `surface_azimuth` in `get_pvgis_hourly` should be change to clockwise from north, but the documentation in [`pvgis.py`](https://github.com/pvlib/pvlib-python/blob/5c20c8c908de97e0e342420528530f3f01352318/pvlib/iotools/pvgis.py#L79-L87) mentions counter-clockwise from north, even though the implementation `'aspect': surface_azimuth-180` matches what was discussed in this issue.
I would have a PR at the ready if this is not a misunderstanding on my side.

AdamRJensen, 2023-08-22:
Hi @tongpu This got fixed in a recent release of pvlib. Please check the What's New section for the specific version.

PR #1739 (2023-05-25): Change surface_azimuth convention in get_pvgis_hourly

 - [x] Closes #1724
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the [... cut]
PR #1838 (2023-08-22): Fix incorrect description of surface_azimuth parameter in get_pvgis_hourly
 - [x] Closes #1724
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~~[ ] Tests added~~
 - ~~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all chang [... cut]