pvlib/pvlib-python #1734: make projected solar zenith a convenience function

mikofski, 2023-05-11:
**Is your feature request related to a problem? Please describe.**
>A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
As mentioned in https://github.com/pvlib/pvlib-python/pull/1725#discussion_r1190000002 the projected solar zenith (PSZ) is used more than once, and may benefit from having a standalone function for reuse to remove redundant code.

**Describe the solution you'd like**
>A clear and concise description of what you want to happen.
a new function perhaps in tools or tracking like this:
```python
sp = pvl.solarposition.get_solarposition(times, latitude, longitude, ...)
proj_zenith = np.degrees(np.arctan2(
    np.sin(np.radians(sp.azimuth))*np.sin(np.radians(sp.apparent_zenith)),
    np.cos(np.radians(sp.apparent_zenith))))
```

**Describe alternatives you've considered**
>A clear and concise description of any alternative solutions or features you've considered.
For linear shade calculation in #1725 the user is required to calculate PSZ externally.

**Additional context**
>Add any other context or screenshots about the feature request here.
one version of the PSZ calculation is currently in `tracking.py` based on the solar vector as seen in the orientation of the tracker axes, that's why axis tilt appears in this "transformed" solar vector:
```python
xp = x*cos_axis_azimuth - y*sin_axis_azimuth
zp = (x*sin_axis_tilt*sin_axis_azimuth
          + y*sin_axis_tilt*cos_axis_azimuth
          + z*cos_axis_tilt)
```
https://github.com/pvlib/pvlib-python/blob/275e6718caf7486cb5b7dcf29acd59499ad51f7f/pvlib/tracking.py#L432

kandersolar, 2023-05-12:
I'm definitely in favor of having a function to calculate this quantity.  It's a step towards https://github.com/pvlib/pvlib-python/discussions/1581#discussioncomment-4060499.

adriesse, 2023-05-20:
This angle is also referred to as the profile angle sometimes, correct?

echedey-ls, 2023-11-04:
> This angle is also referred to as the profile angle sometimes, correct?

I'd like to know that too, in order to contribute to this issue and move forward #1725

mikofski, 2023-11-04:
IDK the correct name for this. If anyone has a stronger opinion please weigh in, otherwise @echedey-ls  just go with “profile angle” as @adriesse suggests. As long as it’s well defined, hopefully that should be enough. Do not let perfect be the enemy of good.

echedey-ls, 2023-11-06:
I don't get how `axis_tilt` and `axis_azimuth` should be implemented here. @kandersolar can you point me to any equation that takes that into account?

wholmgren, 2023-11-16:
> Where should it live? pvlib.tracking and pvlib.shading seem appropriate to me. If this function is intended to be user-facing (which I think it should be), I'd be -1 to putting it in pvlib.tools.

I'm also -1 to a user facing function in `pvlib.tools`. `pvlib.irradiance` is a bit of a junk drawer, but is it more appropriate than the relatively specific `pvlib.tracking` and `pvlib.shading`? At the risk of scope creep and unwise API deprecations, `geometry.py` came to mind for this and the AOI functions in `irradiance`.

echedey-ls, 2023-11-16:
I'm in favour of a `geometry.py` or something along `angles.py` or `angular.py` for this function, `aoi` and `aoi_projection`. I didn't expect the latter two to be in `irradiance.py`

adriesse, 2023-11-16:
For the curious:

https://www.google.com/search?q=%22profile+angle%22+sun

cwhanse, 2023-11-17:
For me, `aoi.py` > `geometry.py`. Into `shading.py` seems like a last resort. My thought is that a user might guess a function like this would be in `aoi.py` and wouldn't know what to expect of `geometry.py`.

adriesse, 2023-11-17:
In case there aren't enough options yet, maybe ```spatial.py```?  I would personally go the path of least resistance and put it in ```irradiance.py``` until the next **big reorg**.

PR #1904 (2024-03-08): Projected zenith convenience function
 - [x] Closes #1734 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with ` [... cut]
closed 2024-03-08 by kandersolar