pvlib/pvlib-python #1735: v0.10.0 release plan

kandersolar, 2023-05-16:
Our next release is scheduled for mid June.  We currently have a [v0.9.6 milestone](https://github.com/pvlib/pvlib-python/milestone/40), but I suggest we merge it into the [v0.10.0 milestone](https://github.com/pvlib/pvlib-python/milestone/34) and release v0.10.0 next so that we can deal with this accumulation of breaking changes:

- #1654
- #1717
- #1719 
- #1724 

Any objections to the next release being v0.10.0?

mikofski, 2023-05-16:
good idea

cwhanse, 2023-05-16:
No objections for me, but let's make a complete list of open issues/PRs to include.

kandersolar, 2023-05-16:
I've relabeled all the v0.9.6 items to v0.10.0, and then took the liberty of moving old items that aren't making progress into the [someday](https://github.com/pvlib/pvlib-python/milestone/3) milestone.  Here is a complete accounting of what remains:

Candidate PRs to merge:

- #1395
- #1547
- #1623
- #1628
- #1654
- #1666
- #1705
- #1707
- #1708
- #1719
- #1725
- #1730
- #1733
- #1736
- #1737

Issues without PRs:
- #1717
- #1724

Needs decision:
- #1687
- #1497

Let's wait to migrate the existing 0.9.6 "what's new" content over to the 0.10.0 file until the finalization PR so that we don't create a bunch of unnecessary merge conflicts in the open PRs.

AdamRJensen, 2023-05-16:
Someone used the ``get_solaranywhere`` iotools function at the PVPMC workshop FYI - pretty cool!

kandersolar, 2023-05-26:
Should we remove `forecast.py` in v0.10.0?  It was deprecated in v0.9.1 (released March 29, 2022).

adriesse, 2023-06-07:
I notice that we're a week from the mid-June target and about half of the proposed PR's are still open.  And mid June coincides with PVSC as well as InterSolar in Europe.  Do we need to adjust 'what' or 'when'?

kandersolar, 2023-06-08:
> Do we need to adjust 'what' or 'when'?

Probably both.  Seems like reviewer capacity is pretty limited at the moment.  I guess it might be slightly nicer to not make a new version with breaking changes right after Mark and Silvana's tutorial at PVSC, but overall I don't mind doing the release in "mid-late June".

kandersolar, 2023-06-21:
@pvlib/pvlib-maintainer Time to get back to reviewing now that people are back from conferences!  I'd like to finalize v0.10.0 next week.

This PR that should not be deferred to 0.10.1 is still in need of review:

- #1700

The following PRs have at least one approving review and I plan to merge them on Friday June 23.  If you think you might object, please review them before then, or at least indicate the desire to review.

- #1296
- #1666
- #1743
- #1766 
- #1769
- #1771

If you want any additional PRs to be included in 0.10.0, please list them below.  Otherwise, I'll plan to push the remaining items (all additive) in the [v0.10.0](https://github.com/pvlib/pvlib-python/milestone/34) milestone to v0.10.1.

- I know @cwhanse wants #1708
- @kandersolar wants #1767

mikofski, 2023-06-24:
Ready for takeoff

cwhanse, 2023-06-27:
Include fixing #1731

PR #1793 (2023-06-30): Finalize 0.10.0
 - [x] Closes #1735
 - [x] Pull request is nearly complete and ready for detailed review.
 - [x] Maintainer: Appropriate GitHub Labels (including `remote-data`) and Milestone are assigned to the Pull Request and linked Issue.

Ready for review.  

Copying this from #1766 so it doesn't get forgotten:

> Removing pvlib.forecast, combined with https://github.com/pvlib/pvlib-python/pull/1654, means that several optional dependencies (siphon, netcdf4, cftime) are no longer used anywhere and can also be removed. This PR removes them here, but we should also remember to remove them fr [... cut]
closed 2023-06-30 by kandersolar