pvlib/pvlib-python #1741: pvlib.pvsystem.retrieve_sam example has no attribute

nativesci, 2023-05-18:
I tried running example code from https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.retrieve_sam.html.  And it throws this error. AttributeError: 'DataFrame' object has no attribute 'AE_Solar_Energy__AE6_0__277V__277V__CEC_2012_'

Although I was just trying to get the example code working, what I really want is to get the solar modules available to choose from.

kandersolar, 2023-05-18:
Hi @nativesci, thanks for pointing out this error in our documentation.  The `CECmod` and `CECInverter` tables (for PV modules and PV inverters, respectively) change over time as the CEC adds, removes, and renames entries.  It seems that the inverter name in that example code doesn't exist anymore in the `CECInverter` table currently included in pvlib.  We should change that example code to use an inverter name that is in the table.  Let's leave this issue open to track that documentation update.

> what I really want is to get the solar modules available to choose from.

If you want PV modules (rather than inverters), you'll need to use `pvsystem.retrieve_sam('CECmod')`.  This returns a table of parameters for the CEC single-diode PV module model, where each module is a column in the table.  That table is stored in a CSV file that you can also inspect with Excel if you like.  It can be downloaded here: https://github.com/pvlib/pvlib-python/blob/main/pvlib/data/sam-library-cec-modules-2019-03-05.csv

nativesci, 2023-05-18:
Thank you. I also want say that I would be happy to contribute to the documentation. Sent from my iPhone

PR #1833 (2023-08-15): Fix example in docstring pvsystem.py
 - [x] Closes #1741
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with [... cut]
closed 2023-08-15 by kandersolar