pvlib/pvlib-python #1745: create a static source of broadband AOD for simplified solis and bird models

mikofski, 2023-05-19:
**Is your feature request related to a problem? Please describe.**
There are not a lot good sources for AOD that are easy to use. Even the ECMWF MACC model is slow and laborious. However, the simplified solis and Bird models require AOD as input. The only clear sky model that works out of the box is Ineichen, but there are some limiting behaviors that users have brought up over the years, especially in very high latitudes (Canada?).

**Describe the solution you'd like**
create a static source of broadband AOD based on the ECMWF MACC data before it disappears forever. Also useful would be the 380nm and 550nm bands for the Bird model. Broadband can be either calculate as it is in Bird, or approximated using 700nm or 750nm.

**Describe alternatives you've considered**
There might be another model, maybe CAMS? Also Aeronet or MODIS?

**Additional context**
[Google group discussion: "Reverse AOD via Linke Turbidity ..."](https://groups.google.com/g/pvlib-python/c/g2-hJtGnR9E/m/AAK5xYPKAgAJ)

adriesse, 2023-05-20:
What spatial and time resolutions would you aim for?

mikofski, 2023-05-20:
IDK, if the default resolution is too big to distribute, I think it’s netCDF4, then we should downscale to a reasonable size? Another idea is to host it as git LFS in its own repo. There was another thread about hosting test data in a separate repo. What resolution are you considering?

adriesse, 2023-05-20:
IDK either.  I'm starting to imagine a global "TMY" of multiple atmospheric parameters with the dimensions time and space.  I guess you just downsample until it reaches a reasonable size.  But some background research would be warranted to see what already exists.

mikofski, 2023-05-21:
There is only 10 years of MACC data. Is that enough to form a typical year? From our paper with Cliff, the atmosphere was surprisingly stable wrt broadband AOD so maybe…?