pvlib/pvlib-python #1747: Is there interest in implementing a .pan or .ond reader into iotools?

ckrening, 2023-05-24:
I find that .pan or .ond files are the most widely/easily available files containing parameters to use with the existing module and inverter models in PVlib. I have created read and parse functions to streamline getting those parameters into my code. 

Would enough people find this feature useful that I should create a pull request?

cwhanse, 2023-05-24:
@ckrening there is significant interest in having this capability. Pull requests welcome :)

kandersolar, 2023-05-24:
Yes!  This has been discussed several times over the years, but nobody has taken the initiative to do it.  Here are some related discussions, for reference:
- PAN and OND readers on the draft roadmap: https://github.com/pvlib/pvlib-python/discussions/1581
- https://github.com/pvlib/pvlib-python/issues/1002#issuecomment-658363265
- https://github.com/pvlib/pvlib-python/issues/1094#issuecomment-1086411784
- https://groups.google.com/g/pvlib-python/c/PDDic0SS6ao/m/Z-WKj7C6BwAJ

mikofski, 2023-05-25:
Any chance you can reuse our port @frivollier https://github.com/frivollier/pvsyst_tools

ckrening, 2023-05-25:
Thank you all for the suggestions! I went ahead and created a pull request #1749 using the code I had already created. I took a different approach than the suggested #1002 and #1094. There are trade-offs for either method but, please review and we can discuss what is best!

PR #1749 (2023-09-13): Create `pvlib.iotools.read_panond` for reading .pan and .ond files

 - [x] Closes #1747 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added - I tried to add something but was having difficulty finding where to store test files to read.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pu [... cut]
closed 2023-09-13 by kandersolar