pvlib/pvlib-python #1753: Back to basics for diffuse fraction functions

adriesse, 2023-05-30:
**Is your feature request related to a problem? Please describe.**
Our collection of diffuse fraction model functions (identified as DNI estimation models) is growing.  Currently none of the functions supports the most basic usage I expect, which is to calculate diffuse fraction from kt.

**Describe the solution you'd like**
A set of diffuse fraction functions that take kt as input (and more kwargs if the model needs it) and produces diffuse fraction as output.
A higher level function could be provided to do the higher level task of decomposing ghi using a kwarg to select the model.

**Describe alternatives you've considered**
Implement my own versions of the functions locally, which I have done.

**Additional context**

#1128
#1179
#1685
#1705 
#1730

cwhanse, 2023-05-30:
@adriesse do you think we factor the diffuse fraction = f(kt) functions out of the various decomposition functions?

adriesse, 2023-05-30:
@cwhanse that's the gist of it.

adriesse, 2023-08-14:
Any interest in the f(kt)?  I'm getting ready work on the Erbs PR.