pvlib/pvlib-python #1756: Functions to consider making private

kandersolar, 2023-05-30:
I noticed we have several functions that don't have pages in our API reference, but are also not private (name doesn't start with `_`):

- [pvlib.iotools.midc.format_index](https://github.com/pvlib/pvlib-python/blob/v0.9.5/pvlib/iotools/midc.py#L105)
- [pvlib.iotools.midc.format_index_raw](https://github.com/pvlib/pvlib-python/blob/v0.9.5/pvlib/iotools/midc.py#L129)
- [pvlib.iotools.srml.format_index](https://github.com/pvlib/pvlib-python/blob/v0.9.5/pvlib/iotools/srml.py#L121)
- [pvlib.iotools.srml.map_columns](https://github.com/pvlib/pvlib-python/blob/v0.9.5/pvlib/iotools/srml.py#L93)
- [pvlib.iotools.surfrad.format_index](https://github.com/pvlib/pvlib-python/blob/v0.9.5/pvlib/iotools/surfrad.py#L162)
- [pvlib.scaling.latlon_to_xy](https://github.com/pvlib/pvlib-python/blob/v0.9.5/pvlib/scaling.py#L149)
- everything in [pvlib.spa](https://github.com/pvlib/pvlib-python/blob/v0.9.5/pvlib/spa.py) except `solar_position`, `transit_sunrise_sunset`, `earthsun_distance`, and `calculate_deltat` 
    - Note that these functions lack real pages but do have stub entries here: https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spa.html

I propose we make all these functions private (add a leading underscore to their names).  I doubt they get any meaningful amount of use, so I wouldn't bother with a deprecation for any of them.

kandersolar, 2023-05-31:
> Only pvlib.spa.calculate_deltat has a hyperlink.

Unless I'm mistaken, the calculation in `calculate_deltat` is not actually part of the SPA, despite living in `spa.py`.  I wonder if it should be moved from `solarposition.py`.  If we did that, then none of the functions in `spa.py` would really be user-facing (they are all supposed to be used by the wrappers in `solarposition.py`) and I guess the entire module could be made private (`pvlib._spa`).

wholmgren, 2023-06-15:
I've gone back and forth on this and ultimately I guess it's ok. It seems unnecessarily hostile to users that actually went to the effort to read source code, and we want more of those users rather than antagonizing the few that we have. I've used some of these functions in my own code, but I'm not the typical user and I can update it if I need to. And I'd like to see spa moved to its own package, but who wants to support that? Which is the point, isn't it?

kandersolar, 2023-06-20:
The SPA-related changes in #1769 are now reverted.  Now it only changes the other functions listed above.  I'd rather improve the SPA docs in a separate PR.

kandersolar, 2023-06-23:
On closer inspection, I realized `scaling.latlon_to_xy` was intended to be public, it just wasn't listed in the sphinx docs.  #1769 now lists it instead of making it private.