pvlib/pvlib-python #1757: bishop88_v_from_i

cwhanse, 2023-05-31:
**Describe the bug**

`bishop88_v_from_i` does not converge to a voltage solution when current is greater than short-circuit, and the reverse bias term is active.

**To Reproduce**

```
    pvlib.singlediode.bishop88_v_from_i(resistance_shunt=1000., resistance_series=1., nNsVth=2.3, photocurrent=8., saturation_current=1e-8, current=8.1, breakdown_factor=0.1)

  File "C:\Users\cliff\anaconda3\lib\site-packages\pvlib\singlediode.py", line 372, in bishop88_v_from_i
    vd = newton(func=lambda x, *a: fi(x, current, *a), x0=v0,

  File "C:\Users\cliff\anaconda3\lib\site-packages\scipy\optimize\_zeros_py.py", line 368, in newton
    raise RuntimeError(msg)

RuntimeError: Failed to converge after 100 iterations, value is nan.
```

**Expected behavior**
Converge to a voltage on the reverse bias voltage portion of the I, V curve.

**Versions:**
 - ``pvlib.__version__``: 0.9.5

cwhanse, 2023-08-23:
The "fail to converge" appears to be intrinsic whenever the reverse bias term is activated (`breakdown_factor` > 0) when using either method `newton` or `brentq`. (Using `brentq` returns the failure message `ValueError: f(a) and f(b) must have different signs`).

The reason is that including the reverse bias term changes the IV curve at negative voltage, from being defined for all voltage, to having a vertical asymptote at -Vbr (`breakdown_voltage`). When using Newton's method (for illustration), the iteration will push the candidate voltage for the zero of the objective function to the left of the asymptote, where the objective function is undefined (due to it raising a negative number to the power `breakdown_exp`).  (The objective function is current solved for the IV curve - current at which voltage is wanted). An illustration is below.
[image]

Attempts to get the optimization to work by optimizing on a transformed voltage have so far been fruitless. I've tried, for example, mapping V from the interval (Vbr, Voc) to all reals with inverse logit, and optimizing on the transformed voltage. The transformation removes the problem caused by the asymptote. For Newton's method, once the solution hits the middle "flat" part of the objective function, the iteration shoots far to the left, and then crawls slowly back towards the zero, but so slowly that the method exceeds the default iterations (100). I haven't been able to come up with a transformation that preserves the zero, is defined for all reals, **and** transforms the 1/x behavior near the asymptote to something that would direct the iteration back close to the zero.

We want a **vectorized** numerical method to solve these reverse bias curves, for use in the proposed `mismatch.py` functions #1781 For this reason I haven't tried to fix use of `brentq`.

**A proposed solution**: sacrifice some performance for reliability and use the Golden mean search (from the LambertW functions) to solve in `bishop88_v_from_i` only where the reverse bias term is active.

adriesse, 2023-08-24:
Could change the breakdown approximation to something non-vertical?

adriesse, 2023-08-24:
Instead of clipping, could you perhaps drop the (no longer significant) exponential and solve remaining terms explicitly where needed (V<0)?

mikofski, 2023-08-24:
Maybe consider consulting pvmismatch? Although it has 2 diodes instead of one & is unfortunately not vectorized it solves the “current given voltage” or “voltage given current” equations including reverse breakdown using Newton here:
https://github.com/SunPower/PVMismatch/blob/b391a98c5ded478a73b3c6a08e8de89024be5156/pvmismatch/pvmismatch_lib/pvcell.py#L278

cwhanse, 2023-08-24:
> Instead of clipping, could you perhaps drop the (no longer significant) exponential and solve remaining terms explicitly where needed (V<0)? (I haven't thought this all the way through...)

Dropping the insignificant forward diode current wouldn't help at negative voltage; the challenge to the iteration is the asymptotic behavior of the reverse diode current.

cwhanse, 2023-08-24:
> Maybe consider consulting pvmismatch? 

Does pvmismatch have an exponential-like model for the reverse bias current (breakdown current)? From what I see, it doesn't account for this current https://github.com/SunPower/PVMismatch/blob/b391a98c5ded478a73b3c6a08e8de89024be5156/pvmismatch/pvmismatch_lib/pvcell.py#L276

mikofski, 2023-08-24:
Hi Cliff, sorry I was mistaken and I revised my comment. No, PVMismatch ignores reverse bias when solving I from V or the reverse. You are correct. See my revised comment above.

cwhanse, 2023-08-24:
> 

Aha. So the reverse bias current only comes in when adding curves, rather than when solving for each curve to be added.

mikofski, 2023-08-24:
Reverse bias is included in the full IV curves that are added together in pvmismatch. It’s just not included when searching for specific points on the IV curve.

adriesse, 2023-08-24:
> Dropping the insignificant forward diode current wouldn't help at negative voltage; the challenge to the iteration is the asymptotic behavior of the reverse diode current.

I meant: solve without reverse breakdown first, as you previously suggested.  Then rework the negative voltages using the model with reverse breakdown added and exponential removed rather than a hard clip.

adriesse, 2023-08-24:
> is no longer Bishop's 1988 model

Most things go out of style at some point.