pvlib/pvlib-python #1762: Incorrect use of absolute airmass inside GTI-DIRINT

adriesse, 2023-06-07:
**Describe the bug**

GTI-DIRINT calculates *absolute* air mass internally and uses it for three things:

- kt'
- disc/dirint
- perez

**Expected behavior**

For kt'and perez *relative* air mass must be used. 

For disc/dirint there may be some doubt, although I would go with the original report reference and use *relative* air mass there too.  There is appears to be no follow-up literature documenting the contrary and pvlib functionality should be traceable.

The pressure argument could/should then be dropped from several functions, including disc.

**Work-around**

Set pressure to 101325.

**Versions:**
 - ``pvlib.__version__``: '0+unknown'

**Additional context**
Add any other context about the problem here.

Chetansahney, 2026-11-15:
I'd like to work on this. I'd keep the scope to the Perez part, since that one is unambiguous inside pvlib: `_gti_dirint_lt_90` passes the pressure-corrected airmass to `get_total_irradiance`, while `perez` documents relative (not pressure-corrected) airmass. kt' and DISC would keep the pressure-corrected airmass, which matches what `disc` and `dirint` do today, so this doesn't touch the open question in #1902 about deviating from Marion.

The change is to compute the relative airmass once, keep `get_absolute_airmass(...)` for `_disc_kn` and `clearness_index_zenith_independent`, and pass the relative one to `get_total_irradiance`. At the default pressure nothing changes. Away from sea level it does, e.g. for the `test_gti_dirint` inputs:

| pressure | DNI at noon, before → after | POA recovered by transposing the output with relative airmass, before → after (target 1000) |
|---|---|---|
| 93193 Pa | 649.0 → 657.9 | 996.0 → 999.7 |
| 70000 Pa | 465.9 → 476.6 | 996.4 → 999.95 |

With the fix, transposing the GTI-DIRINT output with relative airmass gives back the input GTI within the 1 W/m² convergence tolerance. Today it is off by up to about 5 W/m² at 70 kPa.

I have this ready with the `test_gti_dirint` pressure case updated, a round-trip test at two pressures, and a whatsnew entry. Does that scope sound right, or would you rather handle the kt'/DISC airmass together with #1902?

cwhanse, 2026-11-16:
@Chetansahney we don't encourage first-time contributors using AI to open PRs, since in our experience those PRs cost more maintainer's time than doing the PRs ourselves.

Chetansahney, 2026-11-16:
Understood, thanks for letting me know. I won't open a PR for this. The numbers above should still be useful to whoever picks it up.