pvlib/pvlib-python #1801: ClearSky's GHI has a 6 hour shift with ECMWF data

carlosg-m, 2023-07-03:
ECMWF in blue
Pvlib in orange
I've verified my source of ECMWF weather data using Airport METARS as reference, and the data matched.
[image]

kandersolar, 2023-07-03:
Oh, I see this location is actually in Portugal, so Tucson's (`tus`) timezone isn't relevant.  My mistake.  I still suggest checking the timezones though.  In case you haven't seen it, this page in the pvlib docs may be helpful: https://pvlib-python.readthedocs.io/en/stable/user_guide/timetimezones.html

carlosg-m, 2023-07-03:
My apologies for the confusing variables names. The timezone of the date range should be in UTC. 

About the weather data, it should also be in UTC. It comes from Portugal's national meteorological organization (IPMA), and it has several data quality issues, but the base is ECMWF forecasts. In this particular case we chose the most recent forecast for each timestamp (the forecast most similar to the observed value). 

The point here is to fix ECMWF's SSRD variable with Pvlib's irradiance estimation, to remove spikes (as you pointed out) where there should be no radiation. I've only confirmed temperature using METARS and both datasets were extremely similar, do you know any source of data to obtain real solar radiation data, so I can triple check?

[image]

kandersolar, 2023-07-03:
One other possible issue is that the data returned by `Location.get_clearsky` is instantaneous, while perhaps the forecast data are supposed to be 3-hour averages.  If so, then it is important to understand the forecast's labeling convention, e.g. beginning of interval, end of interval, or something else.  That could cause a 1.5 hour "shift" for these 3-hour values.  

Anyway just looking at the plot, I don't think the blue curve makes sense if it is in UTC and in Portugal.  Ignoring the little spike, it starts late and ends too late in the day. 

I don't have any good suggestions for ground station data.  My only idea is the BSRN station in Spain, but it is several hundred km away.  If satellite-based or reanalysis data is good enough for this verification then PVGIS might be useful.  Maybe someone else has suggestions.

AdamRJensen, 2023-07-03:
I would compare it to clear-sky data from McClear (CAMS). You can use the `pvlib.iotools.get_cams` function.

AdamRJensen, 2023-07-12:
Closing this issue as it doesn't seem to be a problem with pvlib. For future user questions, we prefer to post those in the pvlib Google group :)

Thanks for using pvlib!

closed 2023-07-12 by AdamRJensen

carlosg-m, 2023-07-19:
It turns out that the ECMWF's [SSRD](https://codes.ecmwf.int/grib/param-db/?id=169) variable is **accumulated** for each forecast horizon, in this case 3 days. By taking the difference it's possible to account for 3h of the delay I reported before. 
[image]
[image]

However there was still 1 hour and 30 minutes difference between Pvlib and ECMWF this is caused by ECMWF's interval convention as mentioned by @kandersolar. In this particular case the forecast frequency is 3h, and for each 3h interval the value reported corresponds to the center of the interval, for example the SSRD reported at the 12h timestamp corresponds to the "instantaneous value" of the 10h:30m period, very similar to calling Pandas `df.resample('3H')` operation on a higher frequency dataset. This seems to happen to all ECMWF variables that I've analyzed, so to keep the dataset consistent I've forced an 1h and 30 minutes delay in PVlib.
[image]

After converting the units, the result now matches almost perfectly in both versions for periods of clear sky!
[image]

There is still an issue with ECMWF's SSRD that I can't account for, sometimes there are small values greater than zero, that "break" the cumulative sum, and create negative solar radiation. I've created an heuristic to remove these values, however it creates a lot of missing values. I have no idea what these values mean.
[image]

Thank you for the help, and keep up the amazing work with this lib.