pvlib/pvlib-python #1803: Facing always when I called a function of pvlib.iotools.read_tmy3

YutoVego, 2023-07-06:
Hi,

I always cannot call pvlib.iotools.read_tmy3 with "map_variables=True" argument, looking into examples Jupiter notebooks such as https://pvlib-python.readthedocs.io/en/stable/gallery/soiling/plot_greensboro_kimber_soiling.html#sphx-glr-gallery-soiling-plot-greensboro-kimber-soiling-py
Would you please tell me how can I solve this error happening below?
read_tmy3(DATA_DIR / '723170TYA.CSV', coerce_year=1990,map_variables=True)

Ref; [pvlib.read_tmy3](https://pvlib-python.readthedocs.io/en/v0.9.0/generated/pvlib.iotools.read_tmy3.html)
[image]

AdamRJensen, 2023-07-06:
Hi @YutoVego 

The `map_variables` argument is a recent addition to the `read_tmy3` function. Thus, **you need to use pvlib v0.10.1**.

You can check your current version by typing:

    pvlib.__version__

You can install a specific version by typing the following command in an Anaconda prompt or similar:

    pip install pvlib==0.10.1

closed 2023-07-06 by AdamRJensen

YutoVego, 2023-07-06:
Hi @AdamRJensen 
Thank you for your prompt answer.
Noted and now it works, thanks!