pvlib/pvlib-python #1810: first_solar_spectral_correction deprecation points to a non-existent alternative

cwhanse, 2023-07-18:
**Describe the bug**

This [line ](https://github.com/pvlib/pvlib-python/blob/3aa2c4db5b704a5789b1f0a39870c5b14212bafb/pvlib/atmosphere.py#L341) points calls to `pvlib.atmosphere.first_solar_spectral_correction` to `pvlib.spectrum.spectral_factor_firstsolar` which doesn't exist. The deprecation should point to `pvlib.spectrum.mismatch.spectral_factor_firstsolar`

**Versions:**
 - ``pvlib.__version__``: 0.10.1

ai-naymul, 2023-07-19:
Hey I would like to work on this issue, may I?

cwhanse, 2023-07-19:
@ai-naymul yes, please!

kandersolar, 2023-07-19:
It's true that `spectral_factor_firstsolar` lives in `pvlib/spectrum/mismatch.py`, but the spectrum `__init__.py` file imports it into the `pvlib.spectrum` namespace: https://github.com/pvlib/pvlib-python/blob/main/pvlib/spectrum/__init__.py#L7

So the function is indeed accessible with `pvlib.spectrum.spectral_factor_firstsolar`:
```python
In [133]: pvlib.spectrum.spectral_factor_firstsolar
Out[133]: <function pvlib.spectrum.mismatch.spectral_factor_firstsolar(precipitable_water, airmass_absolute, module_type=None, coefficients=None, min_precipitable_water=0.1, max_precipitable_water=8)>
```

This is also the name that is used in the docs: https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spectrum.spectral_factor_firstsolar.html

So I think it makes sense to leave the deprecation message as-is.

cwhanse, 2023-07-19:
OK, my mistake. I don't quite grok how that import works and couldn't find the function when I was looking for it.

closed 2023-07-19 by cwhanse