pvlib/pvlib-python #1815: nan irradiances produced by gti_dirint

AdamRJensen, 2023-07-25:
@adriesse and I have encountered what we think is an undesirable behavior of the `gti_dirint` function. Specifically, the reverse transposition method sometimes returns nan irradiances (e.g., GHI) when AOI>90.

This seems to occur when `kt_prime`>1, calculated on this line:
https://github.com/pvlib/pvlib-python/blob/27bb4e5d3fc16a6b7289bd6c659696334e59c324/pvlib/irradiance.py#L2025

When a `kt_prime`>1 is passed to the line below, the calculated DNI is nan.
https://github.com/pvlib/pvlib-python/blob/27bb4e5d3fc16a6b7289bd6c659696334e59c324/pvlib/irradiance.py#L2109-L2110

I think it's an undesirable behavior to be able for the optimization procedure to be able to choose a `kt_prime` value greater than 1 that later causes the method to fail. Thus my suggestion is to set the upper limit of `kt_prime` to 1:

        kt_prime = clearness_index_zenith_independent(kt, airmass, max_clearness_index=1.0)

In fact, this is already implemented in the `gti_dirint` function.
https://github.com/pvlib/pvlib-python/blob/27bb4e5d3fc16a6b7289bd6c659696334e59c324/pvlib/irradiance.py#L1586-L1587

This small change causes no nans to be returned.

cwhanse, 2023-07-25:
I haven't digested the details, but I can't think of a reason not to do this. Physically, `kt` should be less than 1 unless there's temporary overirradiance due to cloud effects. The optimization is explicit in `gti_dirint` so there's no kind of boundary in a scipy function we'd need to accomodate.

adriesse, 2023-07-26:
I think we should be a bit more careful than that and only clip the value when it is passed to ```_dirint_coeffs```.  Or maybe clip it inside the latter function.