pvlib/pvlib-python #1843: RuntimeWarning: overflow encountered in exp in singlediode.py

ElyWacime, 2023-08-27:
While running a solar energy simulation using pvlib, I encountered a RuntimeWarning of overflow in the exp function within singlediode.py.

Warning Message:
/home/user/.local/lib/python3.10/site-packages/pvlib/singlediode.py:763: RuntimeWarning: overflow encountered in exp
  np.exp((Rs[idx_p] * (IL[idx_p] + I0[idx_p]) + V[idx_p]) /

I have a two-file setup:
refactored_code.py contains all the functions required for the simulation.
APIprojectTADD_refactored.py contains the main logic and calls the functions defined in refactored_code.py.
The warning appears when I run APIprojectTADD_refactored.py.
you can find the files in this repo: https://github.com/ElyWacime/TADD_API

Questions:
Is this warning expected under certain conditions, or could it be a bug?
What steps should I take to debug or resolve this warning?
Environment:
Python version: 3.10
pvlib version: 0.10.0

kandersolar, 2023-08-27:
There are two separate things in the text you showed: a warning, and an error.  The warning (`overflow encountered...`) can likely be ignored; see https://pvlib-python.readthedocs.io/en/stable/user_guide/faq.html#i-got-a-warning-like-runtimewarning-invalid-value-encountered-in-arccos-what-does-it-mean

The error is unexpected for pvlib version 0.10.0 (see #1782 and #1673).  Can you double check the version of pvlib you have installed in the environment where you observe this error?  If it is >=0.10.0, then it is likely something that we want to fix in pvlib.  If so, it would be helpful if you could provide a short piece of code, and whatever data inputs are needed, that we can run ourselves to reproduce the error.

cwhanse, 2023-08-28:
@ElyWacime that warning almost certainly occurred on [this line](https://github.com/pvlib/pvlib-python/blob/0c66e09fd9cce305783ef22fd2fffe7aeeed2dbc/pvlib/singlediode.py#L761). It is also unexpected (and can be ignored) but I'd like to reproduce it. I suspect it results from some inputs (effective irradiance and/or cell temperature) that are far outside physical norms, but perhaps not. Can you post code and data to reproduce the warning also?

ElyWacime, 2023-08-28:
@cwhanse @kandersolar thank you for your responses. I attempted to reproduce the issue with only one configuration for the sake of simplicity so that I could share the code with you. However, the error did not occur in that case. The function causing the issue can be found [here](https://github.com/ElyWacime/TADD_API/blob/main/refactored_code.py#L193), and it's implemented at this [line](https://github.com/ElyWacime/TADD_API/blob/main/APIprojectTADD_refactored.py#L247) in the code. 

And yes, @kandersolar, I double-checked the version of pvlib; it's 0.10.0.

cwhanse, 2023-08-28:
@ElyWacime I'm almost certain that warning comes from this [line](https://github.com/pvlib/pvlib-python/blob/0c66e09fd9cce305783ef22fd2fffe7aeeed2dbc/pvlib/singlediode.py#L763) and it is likely caused by a non-physical value of effective_irradiance, like 10^6. That's the only way I can picture getting an overflow from that exponential. I see you are getting module parameters from the CEC file and they look reasonable.

ElyWacime, 2023-08-28:
@cwhanse yes you're 100% correct i missed closed a parenthesis, wish caused the overflow that caused the error, apologies for wasting you're time i'm still new to coding, i didn't notice it at all.

cwhanse, 2023-08-28:
Glad you found the issue @ElyWacime and glad you find pvlib useful for your work.

closed 2023-08-28 by cwhanse