pvlib/pvlib-python #1848: v0.10.2 release planning

kandersolar, 2023-09-06:
Looking at the PRs currently tagged for [v0.10.2](https://github.com/pvlib/pvlib-python/milestone/42), here is a proposed release plan:

Include in v0.10.2:
- #1708
- #1749
- #1763
- #1781
- #1809
- #1821
- #1822
- #1827
- #1829
- #1832

Defer to the future:

- #1295
- #1497
- #1687
- #1725

The milestone release target is currently set to Sep 15.  Of course it is not a hard deadline, but since 0.10.2 is not a major release (i.e. these PRs are not breaking), we should not feel pressure to include the full list in this release.

As always, comments and PR reviews are very welcome!

cwhanse, 2023-09-06:
I'd like to add closing some small issues:

- #1814 
- #1831 

#1781 requires resolving the issue with convergence of `pvlib.singlediode.bishop88_v_from_i` #1787

echedey-ls, 2023-09-15:
I'd like to help with #1814 ; is it just changing "defauls" to "defaults"? There's a huge amount of comments there.

cwhanse, 2023-09-15:
@echedey-ls please fix the typo. The discussion is about why that value is 9.144

PR #1868 (2023-09-21): Finalize 0.10.2
 - [x] Closes #1848
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or  [... cut]
closed 2023-09-21 by kandersolar