pvlib/pvlib-python #1853: Spectral Modeling + IAM

toddkarin, 2023-09-13:
There have been several efforts to improve pv performance modeling using spectral data. PVlib has already incorporated some of these improvements. I'd like to explore going one step further to incorporating the spectral dependence of IAM.

At PVEL, we have developed a spectral IAM measurement. Here's an example scan on a typical modern module:

[image]

What magnitude of effect should we observe on power modeling? Others have seen relatively large mean absolute errors (~2.8%) due to not taking into account spectral effects [1]. The totalized production will have a lower error because of error cancellation between redder and bluer spectra. Thus the correction is probably much larger for an instantaneous performance ratio than a totalized production.

When I ran energy simulations on an unclipped single axis tracker in las vegas using each of the above spectral IAM data curves (rough way to investigate), I found a 0.3% difference in production between AM1.5 and 430 nm. While this is a small number, this is actually important for how tight finances are for current power plants.

The idea is that we could combine some basic spectrum models with this spectral IAM data and observe the effect on production. This will hopefully lead to results that encourage module manufacturers to optimize for total energy production rather than just STC performance. 

**References**
[1] https://www.sciencedirect.com/science/article/abs/pii/S0038092X19312563
[2] https://github.com/liboisq/WO2PV

cwhanse, 2023-09-13:
@toddkarin in the current workflow, Ee (effective irradiance) is the irradiance that is converted to electrical current. Ee is usually estimated from broadband POA by

Ee = f1 x (IAM(aoi) x POAdirect + POAdiffuse)

where f1 is the "spectral modifier", typically a function of air mass (AM) and sometimes other variables, and IAM is the incidence angle modifier that quantifies the reflection losses applied to direct irradiance, and POAdirect and POAdiffuse are broadband quantities. (There are diffuse reflection models, which I'm omitting for simplicity here).

Are you envisioning that equation could look like

Ee = Integral_lambda g_1(aoi, lambda) x POAdirect + g_2(?) x POAdiffuse d(lambda)

where g_1 and g_2 are functions of aoi and spectral irradiance?

kandersolar, 2023-09-13:
I wonder what physical mechanisms are creating the differences between those curves.  My first thought was the wavelength dependence of index of refraction, but some quick modeling does not show variation in IAM anywhere close to the curves shown above:

```python
aoi = np.linspace(0, 90, 10)
# approximate indices of refraction taken from:
# https://refractiveindex.info/?shelf=3d&book=glass&page=BK7
# https://en.wikipedia.org/wiki/Sellmeier_equation
print(' 400 nm:', pvlib.iam.physical(aoi, n=1.53).round(3))
print('1000 nm:', pvlib.iam.physical(aoi, n=1.507).round(3))
```

Output:
```
 400 nm: [1.    1.    1.    0.998 0.993 0.98  0.946 0.859 0.634 0.   ]
1000 nm: [1.    1.    1.    0.998 0.993 0.98  0.947 0.861 0.636 0.   ]
```

> Others have seen relatively large mean absolute errors (~2.8%) due to not taking into account spectral effects [1]

Here also I'm confused, doesn't section 3.3.3.3 of [1] say that spectral dependence of IAM can be neglected and they chose to ignore it?

adriesse, 2023-09-15:
The main physical reasons are:
- the chance of an electron-hole pair getting collected (vs. recombining) depends on where it is generated
- longer wavelength photons tend to travel further before generating an electron-hole pair
- with light coming in obliquely, the distances change

However, AR coatings are also spectrally/directionally complicated.

I think a purely empirical model wouldn't be too hard to formulate, provided the data are available...  But before putting in the effort, it might be good to do a bit more checking on the energetic impact.

toddkarin, 2023-09-26:
@cwhanse, yes, something like an integral over wavelength and angles.

As for the physical mechanism, the front glass ARC accounts for most of the wavelength dependence. I'd hypothesize the 1050 nm enhancement comes from the low absorption coeff of Si there -- as the angle increases, the path length through the silicon increases and the absorption increases.

[image]

You can check out https://github.com/DuraMAT/pvarc for the front ARC coating theory curves.

cwhanse, 2023-09-28:
The 1050nm "enhancement" - by definition, the IAM is the ratio of transmitted irradiance at AOI to the transmitted irradiance at AOI=0. How does 3% more irradiance get transmitted through the material stack at AOI=50 than at AOI=0? At AOi=0, the path lengths are as short as possible.

toddkarin, 2023-09-28:
To be precise, IAM is that amount of photocurrent generated at an angle divided by that at zero degrees, divided by cosine AOI.

At 50 degrees AOI, using (very approximate) Snell's law, the angle in the silicon is 11 degrees, which would correspond to a 1.8% increase in path length. Since the [absoprtion depth at 1050 nm is about 1 mm ](https://www.pveducation.org/pvcdrom/pn-junctions/absorption-depth) we are in the "linear" part of the absorption vs. depth curve. Thus an increase of 1.8% in path length would be expected to give an increase of ~1.8% in absorption.

So it is not more irradiance getting in, it is more of the irradiance getting absorbed. 

(At least this is my best guess!)