pvlib/pvlib-python #1861: Update default transposition model

AdamRJensen, 2023-09-15:
Having the default transposition model be isotropic seems like a very poor choice. Isotropic models are known to have very large biases and should only be used in very specific use cases (generally not be new users).

I did a brief survey of commercial software and found that the vast majority only offer the Hay-Davirs and Perez model, with the 1990 Perez model seeming to be favored as the default model.

It would seem to make sense to have the same default model used in ``pvlib.modelchain.ModelChain``.

https://github.com/pvlib/pvlib-python/blob/cb4c1355712255c27ec5211a3cd9d93f12518613/pvlib/irradiance.py#L305-L310

AdamRJensen, 2023-09-15:
I suppose this would seen as a "breaking change" and would have to wait until 0.11.0 if we choose to do this.

kandersolar, 2023-09-18:
It would be great if `get_total_irradiance` could still be run with the current minimal set of inputs, even if the default model is changed to something more sophisticated.  #1225 got us halfway there by automatically calculating airmass for Perez, but left ET DNI alone since there is not always a datetime index available to calculate it.  Maybe we could examine the inputs to see if a datetime index is available and only raise the error if not?

AdamRJensen, 2023-10-05:
> It would be great if `get_total_irradiance` could still be run with the current minimal set of inputs, even if the default model is changed to something more sophisticated. ... Maybe we could examine the inputs to see if a datetime index is available and only raise the error if not?

This seems like a good approach to me. I suppose it should be incorporated here:
https://github.com/pvlib/pvlib-python/blob/46851d91bb70e9b5521feb7ec119b057ec3d7f70/pvlib/irradiance.py#L454-L455

 What would be the best way of assessing if the index is datetime? Something like this:

    isinstance(d.index, pd.DatetimeIndex)

kurt-rhee, 2024-01-04:
I would like to throw my vote in the hat for Perez as the default transposition model, about 90% of predictions run in PlantPredict use the Perez model (this is without regard to the source of the meteorological data).  This is an imperfect measure of true usage industry wide, but I think it is a nice data point.

kandersolar, 2024-06-17:
> I suppose this would seen as a "breaking change" and would have to wait until 0.11.0 if we choose to do this.

Seems like we can and should do a deprecation period for this first.  That can be implemented any time, so I've removed the v0.11.0 milestone from this issue.