pvlib/pvlib-python #1867: `bifacial.utils.vf_ground_sky_2d` interprets `max_rows` asymmetrically

kandersolar, 2023-09-20:
**Describe the bug**
`vf_ground_sky_2d` considers one fewer sky wedge on the right side of the central wedge than it does on the left side.

**To Reproduce**
I added a line `print(next_edges)` after here: https://github.com/pvlib/pvlib-python/blob/7608eda81d8302d66be6a686f85f9c3eb5cbe955/pvlib/bifacial/utils.py#L170

Then I ran `_ = vf_ground_sky_2d(surface_tilt=0, gcr=0.5, x=0.5, pitch=1, height=1, max_rows=3)` and got the printed text: `[[[0.01598559 0.04556841 0.18086881 0.48507125 0.18086881 0.04556841]]]`.

Each value in that output describes the VF of a visible sky wedge.  The inputs describe a symmetrical system, so the VFs should be symmetrical too, and they are, except one of them is missing... 0.485 for the wedge directly overhead, with three wedges to the left but only two to the right.

**Expected behavior**
I want the number of sky wedges considered in each direction to be equal.

Maybe this is a fencepost problem, and really what I want is effectively a `max_wedges` parameter instead of `max_rows`?

pasquierjb, 2023-10-17:
k goes from `-max_rows` to `max_rows+1` so the symmetry is for x=0 which is indeed problematic if you intergrate betwen 0 and 1 for a small `max_rows`. However as `max_rows` is set to be high enough to neglect the horizon VF, not sure it will change a lot the results.

[image]

Sar-thak-3, 2024-01-07:
Hi @kandersolar, is this issue still open? I would love to contribute to this issue!

cwhanse, 2026-05-08:
> relative to the "current ground segment" (instead of "current row")

+1 to this. But I would exclude the rows that define the current ground segment, so that `max_rows=1` is the sum of wedges -1, 0, 1.

kandersolar, 2026-09-23:
Closed by #2740

closed 2026-09-23 by kandersolar