pvlib/pvlib-python #1877: Function to fit temperature model to measurements

adriesse, 2023-09-28:
**Is your feature request related to a problem? Please describe.**
Many people struggle with fitting the popular temperature models to field data, or in other words, to determine the model parameters from field data.  

**Describe the solution you'd like**
An effective method of fitting has been described and demonstrated in Jupyter Notebook published by Sandia Labs on [DuraMAT ](https://datahub.duramat.org/dataset/module-temperature) data hub along with an open data set of field data for demonstrating and testing temperature model fitting methods.

It would be great to make this method available through a function in PVLIB so that it can be more easily used. 

**Additional context**
There is fast-growing interest in fitting temperature models because of the unique conditions in floating and agri-PV, therefore this capability is needed sooner rather than later.  The method is also simple enough to be a candidate for IEC-61853-2.

adriesse, 2023-11-02:
This feature is currently blocked.  See #1898.

wholmgren, 2023-11-02:
In my opinion, @williamhobbs engagement with the pull request and the reference material demonstrates that the contribution is 1. useful and 2. sufficiently documented.

adriesse, 2023-11-05:
Just in case it went unnoticed, I uploaded the notebook to Zenodo for archival purposes on October 14, 2023, and updated the references in the doc strings accordingly.

adriesse, 2023-11-05:
> I can relent and not stand in the way of #1878 if there is a commitment to produce a document (journal or conference paper, or technical report) of some kind in the near future.

Does anyone following this thread (or not) have extra time and funding available to work on such a document?

closed 2023-12-04 by adriesse